pulp-platform/pulpino #2: Question about riscv-toolchain

huangchengxtu, 2016-03-03:
Hi
When i use the official RISC-V toolchain supported by Berkeley,always installing riscv64-unknown-elf-gcc,not riscv32-unknown-elf-gcc .what are the differences between them? 
Or what do i need to do?

thanks,wait for you anser!

atraber, 2016-03-03:
Hi

riscv64-unknown-elf-gcc is intended for a 64-bit RISC-V target. Our core however is 32-bit only, so you will need the riscv32-unknown-elf-gcc.

When configuring the RISC-V gnu toolchain, just append `--with-xlen=32`. This should then build the 32-bit version of it which is compatible with our core.

Cheers
Andy

closed 2016-03-05 by atraber

riteish4141, 2018-01-08:
Hello,

Can you please elaborate where to append '--with-xlen=32' ?