pulp-platform/pulpino #3: Fail to make datasheet

Erik-Chen, 2016-03-05:
Hi everyone,
    I want to make datasheet at /doc/datasheet. But there are many file missing and errors, as following.

LaTeX Error: File `algorithm2e.sty' not found.
LaTeX Error: Unknown option `ruled' for package `algorithm2e' 
LaTeX Error: File `bytefield.sty' not found.
........

My environment is Ubuntu 14.04 64-bits.
I think just put a pdf on github is a faster way. Thanks!!!

atraber, 2016-03-05:
Hi

It seems that you are missing the algorithm2e package for LaTeX. It should be easy to find in the packet repositories of ubuntu.

We don't want to upload the compiled pdf file on github as this will bloat the repository and increase it in size (binary files...)

We have uploaded a compiled version on our website: http://www.pulp-platform.org/wp-content/uploads/2016/02/pulpino_datasheet.pdf

closed 2016-03-05 by atraber

Erik-Chen, 2016-03-05:
Thank you very much!!!