pulp-platform/pulpino #6: pulpino/sw/apps/riscv_tests/testCSR/testCSR.c:153: Error: Instruction csrrw requires absolute expression

berndca, 2016-03-19:
I have run into problems building the sw.

I'm getting the error message:

    Assembler messages:
    pulpino/sw/apps/riscv_tests/testCSR/testCSR.c:153: Error: Instruction csrrw requires absolute expression
    pulpino/sw/apps/riscv_tests/testCSR/testCSR.c:153: Error: Instruction csrr requires absolute expression

I'm using https://github.com/riscv/riscv-gnu-toolchainriscv-gnu-toolchain configured with 

    ./configure --prefix=/opt/riscv --with-xlen=32

I did not make any changes to cmake_configure.riscv.gcc.sh 

Please advise.
Thanks, Bernd

atraber, 2016-03-30:
I don't know what the issues is here, but it is definitely related to the inline assembly code that we use to check if our CSR registers are working properly.

I'm currently compiling the latest riscv-gnu-toolchain to see if I can reproduce the issue, since our toolchain compiles this piece of code well and also the riscv-gnu-toolchain version I'm having locally works well.

In any case, if this is the only issue that you are seeing, you can also just ignore it. Compiling testCSR is not vital for PULPino

berndca, 2016-03-30:
Thanks for the reply. Could you please confirm that --with-xlen=32 is the only option required for building the pulpino tool chain?

atraber, 2016-03-30:
Yes, at least it is what we were using

atraber, 2016-03-31:
It seems that the privilege specs have been changed. I checked the latest toolchain and there the mcpuid CSR register does no longer exist and the mimpid and mhartid registers have other addresses. So it is not surprising that this gives compile errors.

I have changed our local tests to still compile, but assume the old values for those registers, but probably we will change them for the future as soon as there is a stable spec available.

closed 2016-04-01 by atraber