pulp-platform/pulpino #8: FPGA bitfile fails to build from source

wachag, 2016-04-06:
Steps to reproduce (Vivado 2015.4):
- Clone pulpino
- Execute update-ips.py
- cd to fpga
- make all
Produces an error:
ERROR: [Synth 8-439] module 'riscv_core' not found [/home/wachag/svn/pulpino/rtl/core_region.sv:171]

atraber, 2016-04-06:
I cannot reproduce the issue on our machines, but I can speculate what the issue might be. There are two options I can think of:

1. It is related to https://github.com/pulp-platform/riscv/commit/e899af439535a9459129ae7129b15264d690ccaf which apparently was a problem for Vivado 2015.4, while it worked perfectly for us on Vivado 2015.1

2. It is a "dummy" error. The FPGA synthesis is done twice in the scripts, and the first time it is expected to fail (surrounded by try-catch block). The first time is used to analyze the RTL hierarchy, so that it can work for the second try.

Please check if the error is still there with the newest version of the RISC-V core. If yes, then it is either option 2, or there is something else broken that I'm not aware of currently

rsivasundaram, 2016-04-16:
In relation to this, I am also using Vivado 2015.4 and I couldn't execute update-ips.py because it says the repository is not found. 

- cd to fpga
- make all produces error: couldn't read file "tcl/ips_inc_dirs.tcl":No such file or directory while executing

closed 2016-05-09 by atraber