pulp-platform/pulpino #9: ETH version compiler

Erik-Chen, 2016-04-07:
Hello
   In the readme, there are two versions of compiler we can use for PULPino. I want to build the ETH version to use the full ISA on PULPino. Where can I get the information? Thank you!
Erik

Erik-Chen, 2016-04-07:
I found it at https://iis.ee.ethz.ch/~haugoug/pulp/

ztbztb12345, 2016-05-03:
Hello
How to register at this website? It needs to log in. Thank you!

Erik-Chen, 2016-05-03:
Hi,
The website is locked now. I am not the officials. I already sent an email to them.

atraber, 2016-05-03:
Hi guys,

We protected the website with a password since it was never intended to be open as it contains our internal resources.
We want to completely open-source our compiler and everything related to it as soon as possible, but as you can imagine this is not so simple since you need to clean up code, add some documentation, ...

You can find our timeline here: http://www.pulp-platform.org/release-plan/

If everything goes according to plan, we can share it within a couple of weeks :-)

closed 2016-05-03 by atraber

ztbztb12345, 2016-05-03:
Thank you very much. ^_^