pulp-platform/pulpino #10: problem of generating elf file

lx324310, 2016-04-25:
Hi:
         when i enter into the dir /sw/build/ and make helloworld . the log shows that:
Scanning dependencies of target string
[  0%] Building C object libs/string_lib/CMakeFiles/string.dir/src/qprintf.c.o
Assembler messages:
Error: Architecture doesn't allow hardfloat ABI
make[3]: *** [libs/string_lib/CMakeFiles/string.dir/src/qprintf.c.o] Error 1
make[2]: *** [libs/string_lib/CMakeFiles/string.dir/all] Error 2
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.dir/rule] Error 2
make: *** [helloworld] Error 2
        Cab you give me a solution of this problem? thank you!

atraber, 2016-04-29:
Hm... difficult to say what the problem is.

Can you try `make helloworld VERBOSE=1`? This will give you more information about the build process.

Which compiler are you using? We do not support floating point, so the error about hardfloat ABI is confusing

PerFidio, 2016-05-09:
Hi! I've the same problem. With helloworld and also with boot_code:
**paglia@paglia-OptiPlex-360:~/pulpino/pulpino/sw/build$ make boot_code VERBOSE=1**
/pulpino/pulpino/sw -B/home/paglia/pulpino/pulpino/sw/build --check-build-system CMakeFiles/Makefile.cmake 0
make -f CMakeFiles/Makefile2 boot_code
make[1]: ingresso nella directory "/home/paglia/pulpino/pulpino/sw/build"
/usr/bin/cmake -H/home/paglia/pulpino/pulpino/sw -B/home/paglia/pulpino/pulpino/sw/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/paglia/pulpino/pulpino/sw/build/CMakeFiles 1
make -f CMakeFiles/Makefile2 apps/boot_code/CMakeFiles/boot_code.dir/all
make[2]: ingresso nella directory "/home/paglia/pulpino/pulpino/sw/build"
make -f libs/sys_lib/CMakeFiles/sys.dir/build.make libs/sys_lib/CMakeFiles/sys.dir/depend
make[3]: ingresso nella directory "/home/paglia/pulpino/pulpino/sw/build"
cd /home/paglia/pulpino/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/paglia/pulpino/pulpino/sw /home/paglia/pulpino/pulpino/sw/libs/sys_lib /home/paglia/pulpino/pulpino/sw/build /home/paglia/pulpino/pulpino/sw/build/libs/sys_lib /home/paglia/pulpino/pulpino/sw/build/libs/sys_lib/CMakeFiles/sys.dir/DependInfo.cmake --color=
make[3]: uscita dalla directory "/home/paglia/pulpino/pulpino/sw/build"
make -f libs/sys_lib/CMakeFiles/sys.dir/build.make libs/sys_lib/CMakeFiles/sys.dir/build
make[3]: ingresso nella directory "/home/paglia/pulpino/pulpino/sw/build"
[  0%] Building C object libs/sys_lib/CMakeFiles/sys.dir/src/exceptions.c.o
cd /home/paglia/pulpino/pulpino/sw/build/libs/sys_lib && /opt/riscv/bin/riscv32-unknown-elf-gcc-6.1.0   -I/home/paglia/pulpino/pulpino/sw/libs/sys_lib/inc -I/home/paglia/pulpino/pulpino/sw/libs/sys_lib/../string_lib/inc  -m32 -march=RV32I -Wa,-march=RV32I -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -fdata-sections -ffunction-sections -fdiagnostics-color=always   -o CMakeFiles/sys.dir/src/exceptions.c.o   -c /home/paglia/pulpino/pulpino/sw/libs/sys_lib/src/exceptions.c
/home/paglia/pulpino/pulpino/sw/libs/sys_lib/src/exceptions.c:18:1: warning: 'interrupt' attribute directive ignored [-Wattributes]
 {
 ^
/home/paglia/pulpino/pulpino/sw/libs/sys_lib/src/exceptions.c:25:1: warning: 'interrupt' attribute directive ignored [-Wattributes]
 {
 ^
Assembler messages:
Error: Architecture doesn't allow hardfloat ABI
libs/sys_lib/CMakeFiles/sys.dir/build.make:62: set di istruzioni per l'obiettivo "libs/sys_lib/CMakeFiles/sys.dir/src/exceptions.c.o" non riuscito
make[3]: *** [libs/sys_lib/CMakeFiles/sys.dir/src/exceptions.c.o] Errore 1
make[3]: uscita dalla directory "/home/paglia/pulpino/pulpino/sw/build"
CMakeFiles/Makefile2:342: set di istruzioni per l'obiettivo "libs/sys_lib/CMakeFiles/sys.dir/all" non riuscito
make[2]: *** [libs/sys_lib/CMakeFiles/sys.dir/all] Errore 2
make[2]: uscita dalla directory "/home/paglia/pulpino/pulpino/sw/build"
CMakeFiles/Makefile2:17971: set di istruzioni per l'obiettivo "apps/boot_code/CMakeFiles/boot_code.dir/rule" non riuscito
make[1]: *** [apps/boot_code/CMakeFiles/boot_code.dir/rule] Errore 2
make[1]: uscita dalla directory "/home/paglia/pulpino/pulpino/sw/build"
Makefile:6382: set di istruzioni per l'obiettivo "boot_code" non riuscito
make: *** [boot_code] Errore 2

Thanks.

gautschimi, 2016-05-09:
Thanks for the input. we have looked at it and apparently there is an issue with the RISC-V compiler in combination with GCC 6.1. we don't know yet what exactly the problem is.

I suggest to use GCC 5.3 until this issue has been solved.

gautschimi, 2016-05-10:
We are using this version of the RISC-V toolchain:
d038d596dc1d8e47ace22ab742cd40c2f22d659e

(it is based on GCC-5.3 and works fine with our core)

PerFidio, 2016-05-10:
The version d038.... use GCC 5.2, however it works fine. Thanks!!

gautschimi, 2016-05-10:
ah yes of course:)
enjoy pulpino then!

closed 2016-05-10 by gautschimi

rsivasundaram, 2016-05-16:
I am still having this issue even after using the commented version of RISC-V toolchain that uses riscv32-unknown-elf-gcc version 5.3.0

edcarstens, 2016-08-17:
I'm having exactly the same issue. What should I do to fix it?

PerFidio, 2016-08-24:
@rsivasundaram  Excuse me, I missed your question.
I don't remember, but checking my notes to rebuild the environment this is all I've found:
$ git clone https://github.com/riscv/riscv-gnu-toolchain.git
$ cd riscv-gnu-toolchain
$ git checkout d038d596dc1d8e47ace22ab742cd40c2f22d659e
$ configure --prefix=/opt/riscv --with-xlen=32
$ sudo make
no others particular operation...

edcarstens, 2016-08-25:
I had this issue because I was getting the riscv toolchain from the wrong repository (ETH). This is what the README.md directed me to. It would help if you would point users to the correct riscv toolchain repository both on the pulpino website and in the README.md file.

gautschimi, 2016-08-25:
which readme are you referring to?

rsivasundaram, 2016-08-25:
Yeah as @edcarstens mentioned, the toolchain I got that worked was located at: https://github.com/pulp-platform/ri5cy_gnu_toolchain

edcarstens, 2016-08-25:
The README.md file here in pulp-platform/pulpino (line 41). Instead of saying "use the official RISC-V toolchain ... or the custom RISC-V toolchain from ETH." it would be better to strongly suggest using the one from pulp-platform/ri5cy_gnu_toolchain as @rsivasundaram mentioned.