pulp-platform/pulpino #11: No rule to make target

rsivasundaram, 2016-04-29:
While in the fpga folder, when I run "make all" it fails at one point and reports the error: 
execvp: make: Not a directory
*** No rule to make target: 'include/config/auto.conf', needed by 'include/config/kernel.release'. Stop

Is there any proposed solution to this issue?

atraber, 2016-04-29:
Indeed it sounds like the missing toolchain is preventing you from compiling Linux for the ZYNQ.

Check the output of `which arm-xilinx-linux-gnueabi-gcc`, as long as it does not output a path, it is missing.

You can get the arm-xilinx-linux-gnueabi-* toolchain from the Xilinx SDK. If you have already installed it, you can probably find it here:
/opt/Xilinx/SDK/2015.1/gnu/arm/lin/bin/arm-xilinx-linux-gnueabi-gcc
if you don't have it installed yet, you can get it from http://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/vivado-design-tools/2015-1.html

rsivasundaram, 2016-04-29:
You were correct in it missing the toolchain, so I went ahead and added that, but now it is telling me 
"Application exception: Not found in path: gmake" in the pulpino/fpga/sw/hsi folder. 
Am I missing something else besides the toolchain in my path?

closed 2016-05-03 by atraber