pulp-platform/pulpino #12: Stalling on bootup

rsivasundaram, 2016-05-03:


atraber, 2016-05-03:
I can only speculate since I'm not a Linux expert. I would guess that they are actually sent, since everything up to this point has worked well. Are you sure that you have a DHCP server on network where your Ethernet cable is connected?
Otherwise you can always just set a static IP and bypass the DHCP process

rsivasundaram, 2016-05-03:
No I don't believe I have a DHCP server on the network. How do I set the static IP?

rsivasundaram, 2016-05-04:
Even after setting a static IP, it successfully sets up IP-config: Complete, but stalls at:
"xemacps e000b000.ethernet: Set clk to 25000000 Hz
 xemacps e000b000.ethernet: link up (100/FULL)"

If i just press enter, it goes to "Welcome to Buildroot buildroot login:" Are there any specified credentials for the login and password or should I just wait at xemacps?

rsivasundaram, 2016-05-05:
Turns out that it was an issue with my minicom not printing out the correct sequence of instructions. Currently looking for a more suitable way to connect to the zedboard.

gautschimi, 2016-05-08:
Hello, I was just wondering if you succeeded in the meantime or if you are still having issues to connect

best,
Michael

rsivasundaram, 2016-05-09:
Hi, so it connects fine. It is just in the boot-up that some commands get jumbled and some words are incoherent and turn into symbols.

gautschimi, 2016-05-09:
can u be a bit more specific? is it also not working in simulation(modelsim)?
what toolchain are you using to compile the applications? could it be an endian issue? (we are using little endian)

rsivasundaram, 2016-05-09:
I'm dealing with the fgpa portion, so I am running it on a zedboard and connecting through it with (minicom -D /dev/ttyACM0 -b 115200 -8 -o). Really, the only current issue I have is ssh'ing into it.

gautschimi, 2016-05-09:
oh you are right. the password is "pulp"

rsivasundaram, 2016-05-15:
The password "pulp" also does not seem to work. I am able to ping the ip addr, but when I try to "ssh root@(ip addr)" and input "pulp" as the password, it is telling me "permission denied, please try again." Is there any other password that you think is correct?

gautschimi, 2016-05-20:
can you try to reset the password with passwd?
or simply add the ssh pub key?

rsivasundaram, 2016-05-22:
Yes you were right. adding the ssh pub key was a better solution.

closed 2016-05-22 by rsivasundaram