pulp-platform/pulpino #13: vcompile_ips.csh missing

holysml, 2016-05-07:
Hi, Pulpino team :
Try to make vcompile after configuration done.
However error out with msg 
"
--> Compiling PULPino Platform... 

/abc/proj/pulpino/vsim/..//vsim/vcompile/vcompile_ips.csh: No such file or directory.
make[3]: *** [CMakeFiles/vcompile] Error 1
make[2]: *** [CMakeFiles/vcompile.dir/all] Error 2
make[1]: *** [CMakeFiles/vcompile.dir/rule] Error 2
make: *** [vcompile] Error 2
"

And found snippet in build_rtl_sim.csh =>
"
IP blocks
source ${PULP_PATH}/vsim/vcompile/vcompile_ips.csh  || exit 1
"

Is **vcompile_ips.csh** missed in release?

gautschimi, 2016-05-08:
Hello

the vcompile_ips.csh script is generated by the script "generate-scripts.py"

after executing this script ( and also update_ips) you should be able to compile everything. let me know if it helps

best
Michael

holysml, 2016-05-08:
Hi, Michael :

When execute ./generate-scripts.py =>

Initialized empty Git repository in /abc/proj/pulpino/ipstools/.git/
The authenticity of host 'iis-git.ee.ethz.ch (129.132.2.24)' can't be
established.
RSA key fingerprint is 5d:42:3d:8c:ea:87:37:54:f3:44:f0:98:1b:30:f4:be.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'iis-git.ee.ethz.ch,129.132.2.24' (RSA) to the
list of known hosts.
git@iis-git.ee.ethz.ch's password:
Permission denied, please try again.

User need to sync with Ethz repository?
May user have passwd of git in Ethz repository?

Regards,
Samuel

holysml, 2016-05-09:
Hi, Michael :

After execute ./update-ips.py, this script will download other IP and generate vcompile_ips.csh.
Issue is gone, please close it.
Thank you.

Regards,
Samuel

gautschimi, 2016-05-09:
Hi Samuel

the readme in the pulpino folder actually describes also the next steps (how to compile and run a simple application etc.)

best,
Michael

closed 2016-05-09 by gautschimi