pulp-platform/pulpino #16: Bootloader using S25FL128S

Erik-Chen, 2016-07-04:
Hi guys,
    I want to simulate the bootloader with S25FL128S Verilog model.
S25FL128S has port CS, SCK, SI, SO, WP, HOLD.
But S25FL128S is in QSPI mode, some of the port will configure as inout port.
The PULPino SPI sperate the input and output port as SDI and SDO.
How do you connect the S25FL128S to PULPino SPI Master ?

Thank you!!!

be4web, 2016-07-04:
Hi,

what you need are pad instances, the following should do the job:

```SystemVerilog
module generic_pad
  (
    input  logic in_i,
    output logic out_o,
    inout  logic pad,
    input  logic en_i
  );

  assign out_o = pad;
  assign pad = en_i ? in_i : 1'bZ;

endmodule
```

Instantiate the generic pad for each signal coming from your SPI model. Then depending on the SPI mode (single, quad receiving or quad transmitting) you need to enable the output.

Just for your information we tested the boot process with Spansion's S25fs256s  flash. Depending on how your model may deviate from ours you may need to adapt the boot code. The readme should clarify the process on how to regenerate the boot code.

Hope this helps.

Regards

Florian

Erik-Chen, 2016-07-04:
Thank you !!!
I will try it.

Best regards
Erik

closed 2016-07-04 by Erik-Chen

aorisa, 2019-04-23:
Hi,
For the inout pad,how can we find the en_i signal(direction control for inout pad)?I don't find the direcion control signal in the apb_spi_Master module.
Thank you!