pulp-platform/pulpino #17: Makefile:13: recipe for target 'ips/xilinx_clock_manager/ip/xilinx_clock_manager.dcp' failed

haniami, 2016-07-25:
When I type 'make all' in the fpga directory, it gives out the following lines? May I know how to encounter this? Thanks in advanced.

/mnt/blockstorage/pulpino-master/fpga# make all
make -C ips/xilinx_clock_manager/ clean all
make[1]: Entering directory '/mnt/blockstorage/pulpino-master/fpga/ips/xilinx_clock_manager'
rm -rf ip/*
mkdir -p ip
rm -rf xilinx_clock_manager.xpr
rm -rf xilinx_clock_manager.hw
rm -rf xilinx_clock_manager.data
rm -rf xilinx_clock_manager.runs
rm -rf xilinx_clock_manager.srcs
rm -rf xilinx_clock_manager.cache
rm -rf component.xml
rm -rf vivado*.jou
rm -rf vivado*.log
rm -rf vivado*.str
rm -rf xgui
rm -rf .Xil
vivado -mode batch -source tcl/run.tcl
make[1]: vivado: Command not found
Makefile:4: recipe for target 'all' failed
make[1]: *** [all] Error 127
make[1]: Leaving directory '/mnt/blockstorage/pulpino-master/fpga/ips/xilinx_clock_manager'
Makefile:13: recipe for target 'ips/xilinx_clock_manager/ip/xilinx_clock_manager.dcp' failed
make: *** [ips/xilinx_clock_manager/ip/xilinx_clock_manager.dcp] Error 2

gautschimi, 2016-07-25:
hello,

are you using vivado? because it seems that it does not recognize it. 
try to add vivado to your path

best

haniami, 2016-07-26:
Thank you for your feedback. Have already added vivado and xilinx to the path. It gives out the following lines:

[hani@putra11 fpga]$ make all
make -C ips/xilinx_clock_manager/ clean all
make[1]: Entering directory  '/mnt/blockstorage/pulpino-master/fpga/ips/xilinx_clock_manager'
rm -rf ip/*
mkdir -p ip
rm -rf xilinx_clock_manager.xpr
rm -rf xilinx_clock_manager.hw
rm -rf xilinx_clock_manager.data
rm -rf xilinx_clock_manager.runs
rm -rf xilinx_clock_manager.srcs
rm -rf xilinx_clock_manager.cache
rm -rf component.xml
rm -rf vivado*.jou
rm -rf vivado*.log
rm -rf vivado*.str
rm -rf xgui
rm -rf .Xil
vivado -mode batch -source tcl/run.tcl
make[1]: execvp: vivado: Permission denied
make[1]: *** [all] Error 127
make[1]: Leaving directory  '/mnt/blockstorage/pulpino-master/fpga/ips/xilinx_clock_manager'
make: ***  [ips/xilinx_clock_manager/ip/xilinx_clock_manager.dcp] Error 2

Thanks for helping.

wsong83, 2016-07-26:
vivado -mode batch -source tcl/run.tcl
make[1]: execvp: vivado: Permission denied
make[1]: ** [all] Error 127

Do you have permission to run vivado?

haniami, 2016-07-29:
I've changed my path and the errors are solved. Thank you for helping. Just one more question, I get the following lines now:

ke[1]: Entering directory '/mnt/blockstorage/pulpino-master/fpga/ips/xilinx_clock_manager'
rm -rf pulpino.xpr
rm -rf pulpino.data
rm -rf pulpino.runs
rm -rf pulpino.srcs
rm -rf pulpino.sim
rm -rf pulpino.cache
rm -rf pulpino.ioplanning
rm -rf pulpino.hw
rm -rf component.xml
rm -rf vivado*
rm -rf .Xil
rm -rf xgui
rm -rf pulpino.edf
rm -rf pulpino_stub.v
rm -rf pulpino_funcsim.v
vivado -mode batch -source tcl/run.tcl

****** Vivado v2015.1 (64-bit)
  **** SW Build 1215546 on Mon Apr 27 19:07:21 MDT 2015
  **** IP Build 1209967 on Tue Apr 21 11:39:20 MDT 2015
    ** Copyright 1986-2015 Xilinx, Inc. All Rights Reserved.

source tcl/run.tcl
# if { ! [info exists ::env(PULP_CORE)] } {
#   set core "RI5CY"
# } else {
#   set core $::env(PULP_CORE)
# }
# puts "Set PULP core to $core"
Set PULP core to RI5CY
# if { ![info exists ::env(XILINX_PART)] } {
#   set ::env(XILINX_PART) "xc7z020clg484-1"
# }
# if { ![info exists ::env(XILINX_BOARD)] } {
#   set ::env(XILINX_BOARD) "em.avnet.com:zynq:zed:c"
# }
# create_project pulpino . -part $::env(XILINX_PART)
# set_property board $::env(XILINX_BOARD) [current_project]
WARNING: [Common 17-599] Property 'board' is deprecated for object type 'project'. Use 'board_part'.
INFO: [Board 49-70] Updating board vlnv to compatible board_part vlnv
# source tcl/ips_inc_dirs.tcl
couldn't read file "tcl/ips_inc_dirs.tcl": no such file or directory
    while executing
"source tcl/ips_inc_dirs.tcl"
    (file "tcl/run.tcl" line 23)
make[1]: *** [all] Error 1
make[1]: Leaving directory '/mnt/blockstorage/pulpino-master/fpga/ips/xilinx_clock_manager'
make: *** [pulpino/pulpino.edf] Error 2

What does this line means and how do I fix it?
couldn't read file "tcl/ips_inc_dirs.tcl": no such file or directory
    while executing

haniami, 2016-07-29:
proo

closed 2016-08-01 by haniami

colin4124, 2016-11-21:
@haniami I occur this problem too. 

`couldn't read file "tcl/ips_inc_dirs.tcl": no such file or directory`

Do you solve it?

gautschimi, 2016-11-21:
run the script in the pulpino home directory:

"update-ips.py" to get the sources of all ips. and "generate-scripts.py" to create the compile scripts (for example the tcl/ips_inc_dirs.tcl)