pulp-platform/pulpino #21: Which ModelSim Edition should we use?

abdullahyildiz, 2016-07-28:
ModelSim PE Student Edition is only available for Windows.

I couldn't find a ModelSim setup file for Linux except ModelSim-Altera Edition. Is it OK to use it?

atraber, 2016-07-28:
Yes, it is sufficient to simulate PULPino, although certain features may not work out-of-the-box (none of which are important for you, I guess)

abdullahyildiz, 2016-07-29:
OK. By the way, which edition are you using?

abdullahyildiz, 2016-08-02:
Is it required to use ModelSim 64-bit?

I get the following error when I type **make vcompile**:

`Failed to open executable /home/abdullah/altera/16.0/modelsim_ase/linuxaloem//../linux_x86_64pe/vlog in execute mode needed for the option -64.`

gautschimi, 2016-08-03:
we have used 64b versions but its not a must. if you don't have a 64b 
version available, just remove the "-64" flag in 
pulpino/vsim/tcl_files/run.tcl

abdullahyildiz, 2016-08-04:
OK. I fixed the problem related to 32-bit version of ModelSim. But I also have to modify **vcompile_tb.sh** at **vsim/vcompile/rtl/** and **CMakeSim.txt** at **sw/apps/** in addition to **.tcl** files at **vsim/tcl_files**.

To simulate the **helloworld** example, I type the following commands (assuming the current working directory is **pulpino**) and at the end get the following error:

```
$ cd vsim/
$ vlib work
$ vmap work
$ cd ../sw/build/apps/helloworld/
$ make helloworld.vsimc
[  0%] Built target sys
[  0%] Built target bench
[  0%] Built target crt0
[  0%] Built target string
[  0%] Built target helloworld.elf
[  0%] Built target helloworld.stim.txt
[  0%] Built target helloworld.links
[  0%] Built target helloworld.slm.cmd
[100%] Running helloworld in ModelSim
Error loading design
make[3]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc] Error 12
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc.dir/all] Error 2
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc.dir/rule] Error 2
make: *** [apps/helloworld/CMakeFiles/helloworld.vsimc.dir/rule] Error 2
$ cat vsim.log
Reading pref.tcl

# 10.4d

# source tcl_files/run.tcl
# vsim -quiet tb -L pulpino_lib -L apb_gpio_lib -L axi_slice_dc_lib -L apb_event_unit_lib -L axi_node_lib -L riscv_lib -L apb_pulpino_lib -L axi_mem_if_DP_lib -L apb_fll_if_lib -L axi_slice_lib -L apb_uart_lib -L apb_spi_master_lib -L apb_timer_lib -L axi2apb_lib -L axi_spi_slave_lib -L apb_i2c_lib -L adv_dbg_if_lib -L axi_spi_master_lib -L core2axi_lib -L apb_node_lib -L apb2per_lib "+nowarnTRAN" "+nowarnTSCALE" "+nowarnTFMPC" "+MEMLOAD=PRELOAD" -t ps -voptargs="+acc -suppress 2103" 
# Start time: 12:07:04 on Aug 04,2016
# ** Error: (vsim-3170) Could not find '/home/abdullah/Desktop/phd/pulpino/vsim/work.tb'.
# Error loading design
# End time: 12:07:05 on Aug 04,2016, Elapsed time: 0:00:01
# Errors: 1, Warnings: 0
[1]+  Done                    vsim
```

There is no top-level file tb within my directory. Could you help me with this?

gautschimi, 2016-08-04:
> OK. I fixed the problem related to 32-bit version of ModelSim. But I
> also have to modify *vcompile_tb.sh* at *vsim/vcompile/rtl/* and
> *CMakeSim.txt* at *sw/apps/* in addition to *.tcl* files at [...]

have you compiled the RTL?

in your build folder, type "make vcompile"

that should build the whole RTL in the vsim/work folder. you can also 
try to launch this command:

|vsim -quiet tb -L pulpino_lib -L apb_gpio_lib -L axi_slice_dc_lib -L apb_event_unit_lib -L axi_node_lib -L riscv_lib -L apb_pulpino_lib -L axi_mem_if_DP_lib -L apb_fll_if_lib -L axi_slice_lib -L apb_uart_lib -L apb_spi_master_lib -L apb_timer_lib -L axi2apb_lib -L axi_spi_slave_lib -L apb_i2c_lib -L adv_dbg_if_lib -L axi_spi_master_lib -L core2axi_lib -L apb_node_lib -L apb2per_lib "+nowarnTRAN" "+nowarnTSCALE" "+nowarnTFMPC" "+MEMLOAD=PRELOAD" -t ps -voptargs="+acc -suppress 2103"|

and check if the different libs are available in modelsim. If the ips 
are not available, they have either not been compiled, or not linked in 
the modelsim.ini
let me know if you have further problems

best,

michael

abdullahyildiz, 2016-08-04:
Sorry, I forgot to run

`make vcompile`

inside the build folder.

This time I get the following error:

```
...
--> PULPino compilation complete! 
--> Compiling work.tb... 
Compiling component:  work.tb 

** Warning: (vlog-7032) The 32-bit glibc RPM does not appear to be installed on this machine.  Calls to gcc may fail.

make[3]: *** [CMakeFiles/vcompile] Error 1
make[2]: *** [CMakeFiles/vcompile.dir/all] Error 2
make[1]: *** [CMakeFiles/vcompile.dir/rule] Error 2
make: *** [vcompile] Error 2
```

I installed **ia32-libs** but I couldn't find the source of the problem.

gurkaynak, 2016-08-05:
This might indeed be a problem of your environment (OS flavor/installed libraries). There are many smaller differences in various Linux flavors which makes it a bit difficult to pin down the problem to a specific issue. For most commercial SW, Redhat (CentOS, Scientific Linux) or SuSe based distributions are recommended, as the tool vendors try these out and qualify for these distributions. Of course it will also work with other distributions, but there might be some more tweaking required. In our case, we are using CentOS 6.x machines and any recent modelsim version (10.x) should work. 

Could you first make sure that modelsim is working properly, i.e. it can compile and simulate a simple System Verilog description.

BTW, we are also about to finish the Verilator support for pulpino and will release it soon.

abdullahyildiz, 2016-08-05:
Thanks for your help. I switched from Ubuntu to CentOS and I can run ModelSim.

Probably I've made a mistake while installing packages on Ubuntu.

closed 2016-08-05 by abdullahyildiz

WaleedOsama, 2017-12-11:
@abdullahyildiz  i am facing your errors exactly , but i am confused , i used ubuntu at first then i switched to Centos , but i used 64 bit one and i faced same errors , so are you using 32 bit Centos ?
and if you do , do i need to make the changes to files you mentioned above to remove -64 flag ?
thanks

gurkaynak, 2017-12-11:
Our setups use 64bit on CentOS 6/7. It would probably also run with 32, and with a bit of tweaking I am sure you can also get it to run on Debian/Ubuntu (many people do). It is just that the scripts and options were not tested on these installations. 

Please make sure that you can run simple System Verilog programs first, before starting to simulate the entire code base.

abdullahyildiz, 2017-12-12:
@WaleedOsama I used CentOS-6.8-x86_64 and ModelSimSetup-16.0.0.211.

nvlinh87, 2019-01-19:
@abdullahyildiz which modelsim are you using? free or with lience? And ehich gcc version? I'm trying to setup env like yours. Os centos 6.8 and ModelSim 16.0 but I don't have lience yet

gurkaynak, 2019-01-19:
> @nvlinh87 I used CentOS-6.8-x86_64 and ModelSim 32-bit
> (ModelSimSetup-16.0.0.211-linux). This version of ModelSim should run
> with a free license.

I think this is the Altera (or Intel) version of Model/Questasim that 
comes packaged with an FPGA board. I am not sure it would work with. I 
think the version refers to the Quartus release and not the Mentor 
Questasim release.

AFAIK, this also would need a license, and if indeed it has a free 
license I am not sure if you can simulate PULPino with a limited version 
of Model/Questasim. I know there were some student editions bit they had 
some limitations.

Our descriptions are for standard versions of Mentor Questasim.

Hope that helps
KGF

nvlinh87, 2019-01-19:
Let's me try with free lience first. I read the spec of free ModelSim of Intel and it said that it only works with design less than 10K code lines but I count Pulpino with RI5Cy code line and the result is over 20K code lines... Anyway, how about gcc version you are using?

gurkaynak, 2019-01-19:
We would be happy if you could try it out and report how it goes, but it 
is not fair to later say it does not work please fix it… just saying

For the GCC especially if you want xPULP extensions of RI5CY, please 
consider using the SDK, it should pull the correct GCC and add the 
patches for extensions. If you want to do everything yourself, and just 
need standard RISC-V only (no extensions), I think any GCC after 7.1 
would work.

KGF

nvlinh87, 2019-01-20:
I can compile Pulpino with free version of Altera Modelsim Starter Edition.
However, there are some critical warnings like:
** Warning: /home/linh/pulp1/pulpino/vsim/..//ips/riscv/riscv_controller.sv(864): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
I'm not sure whether such warning will affect to simulation process or not..

gurkaynak, 2019-01-20:
As mentioned, you need Questasim, not Modelsim..

KGF