pulp-platform/pulpino #24: How to use spiload to dump stdout of a program run on PULPino

abdullahyildiz, 2016-08-24:
In [README.md](https://github.com/pulp-platform/pulpino/tree/master/fpga#stdout-via-printf-on-pulpino) file under pulpino/fpga directory it says that

When PULPino is run on the FPGA, it transfers all output via printf via UART to the ARM host processor in the ZYNQ. To display it, either use a console program like minicom to read directly from the serial port, or specify a timeout when using `spiload`. `spiload` will connect to the serial port and display everything that PULPino sends via UART until the timeout expires.

When I specify a timeout for helloworld program, no `Hello World!!!!!` string is displayed on the console.

`./spiload -t10 helloworld.stim.txt`

gautschimi, 2016-08-30:
hm I think the correct syntax is: "./spiload --timeout=10 helloworld.stim.txt"

anyway, I've just commited a fix to your ssh problem.

if you set up the ssh connection (add your key on the board) you can also execute your programs directly on the fpga by using the command:
make {testname}.fpga

abdullahyildiz, 2016-08-31:
> ./spiload --timeout=10 helloworld.stim.txt

It didn't work either.

gautschimi, 2016-08-31:
I had a look, and indeed when you are connected over uart, this problem occurs. it's due to the fact that helloworld is a very short program and the console is not yet free when the printf is sent.
I'll push a fix in the spiload which should resolve your issue.

closed 2016-08-31 by gautschimi

abdullahyildiz, 2016-08-31:
It is not a problem for me. I just wanted to test whether my setup is working.