pulp-platform/pulpino #25: Cannot connect to the network at startup

abdullahyildiz, 2016-08-24:
I manually start the network as follows:

```
ifconfig eth0 hw ether 02:01:02:03:04:08 #just assigned random 6 octets
ifconfig eth0 192.168.1.80 up
```

However, I cannot use ssh service. Restarting sshd manually didn't work.

gautschimi, 2016-08-25:
hm can you give some more information?
what have you done already? can you connect over uart?
what is not working when you try to ssh?

abdullahyildiz, 2016-08-25:
I can connect over UART (/dev/ttyPS1) to ZedBoard.

The following messages are displayed while the system is booting:

> Sending DHCP and RARP requests ...... timed out!
> IP-Config: Reopening network devices...
> Sending DHCP and RARP requests ...... timed out!
> IP-Config: Auto-configuration of network failed

I restart sshd manually and then run from my computer

`ssh root@192.168.1.80`

However, nothing happens

gautschimi, 2016-08-30:
there was indeed an issue in the Ethernet setup. the Ethernet frequency was somehow corrupted, I've just pushed a fix. if you rebuild the whole fpga it should work again. if you want i can also share the boot.bin etc.

best,
michael

abdullahyildiz, 2016-08-31:
OK. It worked. Thanks for your help.

closed 2016-08-31 by abdullahyildiz

abdullahyildiz, 2016-08-31:
By the way, what is the root password? Password 'pulp' did not work while trying to connect over ssh.