pulp-platform/pulpino #26: Problems compiling mem_dpi.c (dpiheader.h not found)

edcarstens, 2016-08-24:
I am trying to compile mem_dpi.c in your pulpino/tb/mem_dpi directory to use the DPI (direct programming interface) with the Cadence Incisiv simulator. When I try the following I get an error:

 _gcc -fPIC -shared -o libdpi.so mem_dpi.c -I $CDS_INCISIV_DIR/tools/inca/include_

mem_dpi.c:11:23: fatal error: dpiheader.h: No such file or directory
 #include "dpiheader.h"
                                 ^

I could not find this file anywhere.

gautschimi, 2016-08-25:
this file should be in the pulpino repository:

pulpino/tb/jtag_dpi/dpheader.h

it is compiled by the file pulpino/vsim/vcompile/rtl/vcompile_tb.tcl

best,
michael

atraber, 2016-08-25:
Let me try and clarify this a bit.

the dpiheader.h file is auto-generated by ModelSim, see vsim/vcompile/rtl/vcompile_tb.sh,54:
  vlog -quiet -sv -work ${LIB_PATH} +incdir+${TB_PATH} -dpiheader ${TB_PATH}/jtag_dpi/dpiheader.h ${TB_PATH}/jtag_dpi.sv || goto error

ModelSim uses the jtag_dpi.sv SystemVerilog file as input to generate this header file. In the SystemVerilog file one specifies which functions are imported/exported, together with a prototype. This is interpreted by vlog and converted into a c-compatible format => the dpiheader.h

The dpiheader is then included in the C file that actually provides those functions. By including the declarations that were generated by ModelSim, one ensures that the functions do match the ones imported/exported by SystemVerilog.

The dpiheader is not actually needed for this to work correctly, it is merely a check for a user to ensure that he did not make any errors.
I don't know how Incisive handles this, but I would expect that there is a similar solution to auto-generate dpi headers based on SystemVerilog files.

edcarstens, 2016-08-25:
I was able to generate dpiheader.h from the Cadence simulator, but it was not compatible with yours, so I modified it so the type name matched (mem_packet_t). Then I changed the property names to match (we, addr, size). Once I did that it generated the shared object, libdpi.so. Then I pointed the Cadence simulator to this libdpi.so with the sv_lib option. It simulates, but now there is a problem with $readmemh not finding the SLM files, so because this doesn't happen I get a bunch of illegal instruction exceptions from the processor. (I'll look into this some more and if I can't figure it out or if there is some incompatibility I'll create a new issue for it.)

By the way, since the dpiheader.h file is generated by the simulator it is not part of the repository. It is explicitly being excluded by your .gitignore file. This probably makes sense, but it sure is a stumbling block for people like me trying to use simulators different from modelsim.

edcarstens, 2016-08-25:
FYI, the Cadence-generated dpiheader.h names the struct type "tb__mem_packet_t" as follows:

typedef struct {
  char _a0;
  int _a1;
  int _a2;
} tb__mem_packet_t;

This does not match the one in mem_dpi.svh:

typedef struct {
  byte we;
  int addr;
  int size;
} mem_packet_t;

I assume "char" is the same as "byte" type.

closed 2016-08-25 by edcarstens

atraber, 2016-08-25:
Thanks a lot for looking into this! Easiest would probably be to just support cadence incisive from the start, since you are definitely not the only person using it.

Regarding the slm files, you need to create those using the software build scripts. As you probably noticed already, they contain the instructions for the core, thus without them, the core is stuck executing illegal instructions.