pulp-platform/pulpino #27: stdout of PULPino

abdullahyildiz, 2016-08-25:
Which pins of ZedBoard are used for stdout of PULPino. It seems you've used EMIO for UART0 but I couldn't find a location constraint within Vivado project of pulpemu.

abdullahyildiz, 2016-08-25:
I think there is no such pin going outside from ZedBoard, ttyPS0 which corresponds to UART0 is connected to PULPino on programmable logic.

gautschimi, 2016-08-30:
stdout goes over uart to the ps (so internal). it will be pasted on the console which means you should see it when you login to the ps with minicom or ssh.

abdullahyildiz, 2016-09-01:
What is the baudrate of UART0 which is connected over /dev/ttyPS0?

closed 2016-09-01 by abdullahyildiz

z419379295, 2017-03-17:
Hi,

I also have this question, what is the baud rate of UART0 which is connected over /dev/ttyPS0? If it is not standard baud rate, how can we observe the output of PULPino using Minicom? It misses some characters.

Best,
Ken