pulp-platform/pulpino #28: Open failed on file "stdout/uart". No such file or directory

edcarstens, 2016-08-25:
When I simulate pulpino using the Cadence (irun) simulator, I get this error:

_Open failed on file "stdout/uart". No such file or directory_

I found the offending code in uart_tb_rx.hdl.sv:

file = $fopen("stdout/uart", "w");

Should I create a directory called stdout or is this supposed to direct output to the console (standard out)? If so, why not just use $display or $write to do this?

gautschimi, 2016-08-25:
it is writing to a file (sdtout/uart) in your application build folder:

.../pulpino/sw/build/apps/helloworld/stdout/uart

the file doesnt exist in the beginning and should be created by $fopen 
(actually it is deleted before executing the application)

however, if the folder stdout is created by the makefile 
(apps/CMakeSim.txt )

edcarstens, 2016-08-25:
Okay, I will create the stdout directory in my run directory. Thanks.

closed 2016-08-25 by edcarstens