pulp-platform/pulpino #30: Problems simulating pulpino tests with Cadence simulator

edcarstens, 2016-08-29:
I am using the Cadence simulator, irun, to run one of the tests in the pulpino test suite: apps/riscv_tests/testMacNorm. The code seems to compile okay as is, but causes the simulator to hang. From this state, the simulator cannot be used anymore and the job must be killed.

What I found, however, is that if the optimization level is changed (from 3 or 2) to 1 for just the run_benchmark and run_suite subroutines in bench.c, and the main function in testMacNorm.c is modified so it does not ever return, and the testcases in testMacNorm are modified to only include the defined testcases, then the tests complete successfully and the simulator can be stopped (it does not hang).

I also wonder how these tests are supposed to flag your simulator (Modelsim) to finish. since I did not find any $finish in the testbench files.

atraber, 2016-08-29:
Regarding the behavior of ncsim, i.e. that it hangs.
I experienced the same issue when running PULPino with it. The reason for this turned out to be deep in the simulator internals and due to a (in my opinion) strange behavior of the simulator.

In contrast to ModelSim, ncsim may trigger always verilog blocks directly, without opening another delta cycle, when certain conditions are met. In those cases, one may enter an endless loop where two always blocks keep calling each other, although no signal actually changes its value. Cadence calls this zero-width glitches.
To fix this, use the option "-delay_trigger" on irun. The cadence help topic that explains delay_trigger also contains an explanation when this may happen.

Regarding how the simulation stops, there is a $stop in the tb.sv file. After the stop, a tcl script is executed that reads a signal value in the testbench which ModelSim uses to determine its own exit status, so that it can be run by continuous integration frameworks.

edcarstens, 2016-08-29:
The -delay_trigger option solved the issues! I no longer have to change optimization level in bench.c and I can exit main() so the simulation stops (but does not hang). I also verified that leaving undefined tests in the testcases list (the way the code was originally) does not cause a simulator hang anymore.

closed 2016-08-29 by edcarstens

origintfj, 2017-05-19:
@atraber I am using the Cadence simulator, thanks for pointing out the existence of that flag.  Adding it did indeed solve the issue as @edcarstens reported.

If you're interested to know what the root cause of the issue is, I've tracked it down to the stall control always_comb block in controller.sv.

Splitting this always_comb:

```
  always_comb
  begin
    load_stall_o   = 1'b0;
    jr_stall_o     = 1'b0;
    deassert_we_o  = 1'b0;

    // deassert WE when the core is not decoding instructions
    if (~is_decoding_o)
      deassert_we_o = 1'b1;

    // deassert WE in case of illegal instruction
    if (illegal_insn_i)
      deassert_we_o = 1'b1;

    // Stall because of load operation
    if ((data_req_ex_i == 1'b1) && (regfile_we_ex_i == 1'b1) &&
        ((reg_d_ex_is_reg_a_i == 1'b1) || (reg_d_ex_is_reg_b_i == 1'b1) || (reg_d_ex_is_reg_c_i == 1'b1)) )
    begin
      deassert_we_o   = 1'b1;
      load_stall_o    = 1'b1;
    end

    // Stall because of jr path
    // - always stall if a result is to be forwarded to the PC
    // we don't care about in which state the ctrl_fsm is as we deassert_we
    // anyway when we are not in DECODE
    if ((jump_in_dec_i == BRANCH_JALR) &&
        (((regfile_we_wb_i == 1'b1) && (reg_d_wb_is_reg_a_i == 1'b1)) ||
         ((regfile_we_ex_i == 1'b1) && (reg_d_ex_is_reg_a_i == 1'b1)) ||
         ((regfile_alu_we_fw_i == 1'b1) && (reg_d_alu_is_reg_a_i == 1'b1))) )
    begin
      jr_stall_o      = 1'b1;
      deassert_we_o     = 1'b1;
    end
  end
```
Into these two:
```
  always_comb
  begin
    load_stall_o   = 1'b0;
    deassert_we_o  = 1'b0;

    // deassert WE when the core is not decoding instructions
    if (~is_decoding_o)
      deassert_we_o = 1'b1;

    // deassert WE in case of illegal instruction
    if (illegal_insn_i)
      deassert_we_o = 1'b1;

    // Stall because of load operation
    if ((data_req_ex_i == 1'b1) && (regfile_we_ex_i == 1'b1) &&
        ((reg_d_ex_is_reg_a_i == 1'b1) || (reg_d_ex_is_reg_b_i == 1'b1) || (reg_d_ex_is_reg_c_i == 1'b1)) )
    begin
      deassert_we_o   = 1'b1;
      load_stall_o    = 1'b1;
    end

    // Stall because of jr path
    // - always stall if a result is to be forwarded to the PC
    // we don't care about in which state the ctrl_fsm is as we deassert_we
    // anyway when we are not in DECODE
    if ((jump_in_dec_i == BRANCH_JALR) &&
        (((regfile_we_wb_i == 1'b1) && (reg_d_wb_is_reg_a_i == 1'b1)) ||
         ((regfile_we_ex_i == 1'b1) && (reg_d_ex_is_reg_a_i == 1'b1)) ||
         ((regfile_alu_we_fw_i == 1'b1) && (reg_d_alu_is_reg_a_i == 1'b1))) )
    begin
      deassert_we_o     = 1'b1;
    end
  end

  always_comb
  begin
    jr_stall_o     = 1'b0;
    // Stall because of jr path
    // - always stall if a result is to be forwarded to the PC
    // we don't care about in which state the ctrl_fsm is as we deassert_we
    // anyway when we are not in DECODE
    if ((jump_in_dec_i == BRANCH_JALR) &&
        (((regfile_we_wb_i == 1'b1) && (reg_d_wb_is_reg_a_i == 1'b1)) ||
         ((regfile_we_ex_i == 1'b1) && (reg_d_ex_is_reg_a_i == 1'b1)) ||
         ((regfile_alu_we_fw_i == 1'b1) && (reg_d_alu_is_reg_a_i == 1'b1))) )
    begin
      jr_stall_o      = 1'b1;
    end
  end
```
allows simulation to run without hanging without having to apply the '-delay_trigger' option.  The reason for this is as follows:

The SV module shown below causes the Cadence simulator to spin in an infinite loop, even though there isn't actually a combinational loop in the logic.

```
//`define DONT_FAIL

module test
    (
    );

    // proc 1
    logic a;
    logic c;
    // proc 2
    logic b;

    // proc 1
    //
    always_comb
    begin
        // unrelated to the below
        a = 1'b0;
        a = 1'b1;
`ifdef DONT_FAIL
    end
    // split into two seperate processes
    always_comb
    begin
`endif
        // unrelated to the above
        c = b;
    end

    // proc 2
    //
    always_comb
    begin
        b = 1'b0;
        b = a;
    end
endmodule
```
The reason for the spin is as follows:

1. proc1 is evaluated by the simulator and 'a' is assigned, adding it to the event queue.
2. proc2 is evaluated and 'b' is assigned, adding it to the event queue.
3. proc1 is evaluated as a result of the event on 'b' and 'a' is assigned, adding it to the event queue.
4. proc2 is evaluated as a result of the event on 'a' and 'b' is assigned, adding it to the event queue.
5. GOTO (3) // now we're sitting in an infinite loop.

The two assignments to each of 'a' and 'b' are crucial in exposing this behaviour.  If at any time during the execution of the always block the value of an assigned variable changes, the variable will be added to the event queue, even if the execution of the always block as a whole does not result in the value of the variable changing.

Assigning default values to signals at the top of an always_comb process and reassigning them later on is standard practice for me when implementing next-state-logic, so the only other way I can see to avoid this through code changes would be to split up the proc1 always_comb block into two, one for the 'a' assignments, and one for the 'b' assignments, and while this may seem an obvious choice in this case, it's unclear to me that it will always be obvious.

The '-delay_trigger' tells the simulator to wait until it's evaluated the entire always block before deciding on whether or not there has been an event on any particular variable, and thus whether or not it should be added to the event queue.  Setting this flag prevents the simulator from hanging in this case because variables which don't change after a complete pass of the always block are not added to the event queue.