pulp-platform/pulpino #31: fft test not working -- .bss will not fit in region 'dataram'

edcarstens, 2016-08-29:
I added a line to include stdlib.h to get malloc (which was not being declared anywhere). But then 'dataram' overflowed by 3744 bytes. Is there a tinier footprint library missing?

gautschimi, 2016-08-30:
how did you compile the fft? did you change the memory sizes? when i compile it, it is far from filling up the data ram. besides, having more functions should only increase the pressure on the instruction memory and not the data memory.

edcarstens, 2016-08-30:
I am not using your cmake environment, so it looks like I probably omitted a C define. Let me look into that.

edcarstens, 2016-08-30:
I did not set KISS_FFT_USE_ALLOCA or FIXED_POINT=16. How do I set these C defines at the GCC command line?

edcarstens, 2016-08-30:
Actually it looks like you have removed fft from the build list (because region 'ram' overflowed by 3572 bytes). See **apps/sequential_tests/CmakeLists.txt**

gautschimi, 2016-08-31:
Ahh I see. You are using another FFT, I was looking at the FFT in:
https://github.com/pulp-platform/pulpino/tree/master/sw/apps/bench/fft

We have removed the one you were using because it was taking too much space. (it is running on our big pulp)

if you need more memory you could also increase it. you have to change:
1) instantiate more memories (rtl, or on the fpga)
2) linker script

closed 2016-08-31 by gautschimi