pulp-platform/pulpino #32: apps/imperio_tests/testI2C ncsim assertion failed

edcarstens, 2016-08-31:
I am running apps/imperio_tests/testI2C on the Cadence simulator. I am seeing the following assertion fail in my log output:

ncsim: *E,ASRTST (/projects/.../riscv_core/generic/riscv_id_stage.icm.sv,1197): (time 19076220 NS) Assertion tb.top_i.core_region_i.RISCV_CORE.id_stage_i.__assert_1 has failed

Branch decision is X
.
(these two lines keep repeating)
.
.
.

edcarstens, 2016-08-31:
I see the same assertion failing for imperio/testSPIMaster too. It fails 6 times, then prints ID: and a bunch of blank lines. The simulation didn't look like it would finish so I interrupted it.

edcarstens, 2016-08-31:
I traced the cause of the X on branch_decision_i to a read of sp ram reading hi-Z data:

regfile_wdata_wb_i[31:0] = 32'bz
core_region_i.core_data_rdata[31:0] = 32'bz
core_region_i.data_mem_sp_ram_i.addr[12:0] = 0x0653

edcarstens, 2016-08-31:
The previous comment relates to the imperio/testSPIMaster test.

gautschimi, 2016-08-31:
this is a typical problem when reading from uninitialized memory.
if the memory is not initialized the model will return xxxx. if you 
perform comparisons on these data you will get this assertion

best,
michael

edcarstens, 2016-08-31:
I don't see imperio/testSPIMaster intializing any memory other than the program code. Does it simulate okay for you? I put some pulldowns on top_i.spi_master_sdi0_i - top_i.spi_master_sdi3_i. Now the simulator log says the test fails and the content of flash memory is all zeros instead of expected non-zero data. So it looks like the test is expecting Flash memory to be initialized. Does it expect a Flash to be connected to the SPI Master?

gautschimi, 2016-09-01:
Ahh I see. Imperio is our ASIC target which in addition contains a 
padframe etc.
we have also connected a flash model to the spi master. but we cant open 
source that one, which means this test will only pass if you connect 
something to the spimaster.

edcarstens, 2016-09-01:
Thanks. Then I shouldn't expect the I2C imperio test to pass (unless I2C is hooked up to something) either, right?

gautschimi, 2016-09-02:
correct, all imperio tests don't work unless you attach something to the ports.

closed 2016-09-02 by edcarstens