pulp-platform/pulpino #33: pulpino subrepos - why not use git submodules?

morse-michael, 2016-09-02:
Hello, 

I was wondering what the reason is for using ups_list.yml, update-ips.py, push-ips.py, etc rather than using git submodules? Using git submodules seems much more robust, tracks the head of subrepos and lets you have subrepos on different remotes. But I suppose there must be a good reason why you're not using git submodules, just wondering what it is...

Thanks,

Michael

gautschimi, 2016-09-02:
Actually we considered using submodules. however, they proved to be impractical for our case which is why we switched to separate repositories. the main reason is that we want to keep the version of the modules changing which is practically impossible with submodules (during development we want to use the head of each module, unless otherwise specified). once we approach a tapeout we tag everything of course to keep the state.

morse-michael, 2016-09-03:
Thanks for explaining @gautschimi, makes perfect sense. I personally prefer to control / keep track of any core level changes from top, but agree that it does cause quite a bit of overhead in the top level repo...

closed 2016-09-03 by morse-michael