pulp-platform/pulpino #35: TimerA ISR err?

ictzhanglei, 2016-09-22:
In "sw/apps/freertos/Source/portable/GCC/RI5CY/port.c +327 +341"
The ISR_TA_CMP should be changed to ISR_TA_OVF, as the irq to CPU is overflow instead of timer comparison?

be4web, 2016-09-22:
Hi,

actually the interrupt should be triggered by comparison. The idea is that the user can determine the time he wants to spend on the tasks.

Timer A output compare register is set in `prvSetupTimerInterrupt `. Are you experiencing problems? Could be that various changes broke the support for the port.

Cheers,

Florian

ictzhanglei, 2016-09-22:
Yes. When I run freertos, the simulator said timer A overflow interrupt occurred, while in the sw code no overflow ISR is registered.

be4web, 2016-09-22:
Indeed there is a problem with the preemptive scheduler I have no quick fix at the moment - the bug was introduced when the way interrupts are handled has changed. My suggestion for the moment would be to switch to the **cooperative scheduler** by setting `#define configUSE_PREEMPTION		 0` to 0 in the `FreeRTOSConfig.h`. 

But I don't see the problem you are mentioning e.g.: the wrong ISR is called. The port is still using the compare ISR. What do you mean by "the simulator says"? Have you traced the waveforms? I personally don't thing that the wrong timer interrupt is called but that you rather run into the problem I've described above.

Sorry for that, I'll try to come up with a fix.

Cheers,

Florian

ictzhanglei, 2016-09-23:
Thanks.

Here is what I got when run simulation in Modelsim (my simulator :-)
[--start--]
RX string: Task 2
RX string: Task 1
           2507720000: Entering exception routine. Cause(3d), PC_MUX(3)
           2522520000: Entering exception routine. Cause(3d), PC_MUX(3)
           2537040000: Entering exception routine. Cause(3d), PC_MUX(3)
           2551840000: Entering exception routine. Cause(3d), PC_MUX(3)
           2566360000: Entering exception routine. Cause(3d), PC_MUX(3)
           2593600000: Entering exception routine. Cause(3d), PC_MUX(3)
Branch decision is X
[--end--]

You can see I have got interrupt timer A ovf  according to the Cause is 3d.

be4web, 2016-09-23:
Interesting output, I can't really reproduce it. Just to be sure: Are you on the latest commit?

ictzhanglei, 2016-09-23:
Yes, I think I am on the latest master commit. I added some assert in the cpu design file, that shows more info when interrupt or exception happens.

closed 2017-09-14 by FrancescoConti