pulp-platform/pulpino #36: hi,,, vcompile error

samiasd, 2016-10-04:
Dear PULPino team,
I'm working to start with PULPino and I'm stuck with the following error when I executed make vcompile

~/pulpino/build$ make vcompile
Scanning dependencies of target vcompile
env: ./vcompile/build_rtl_sim.csh: No such file or directory
make[3]: *** [CMakeFiles/vcompile] Error 127
make[2]: *** [CMakeFiles/vcompile.dir/all] Error 2
make[1]: *** [CMakeFiles/vcompile.dir/rule] Error 2
make: *** [vcompile] Error 2

any suggestion please??
Regards

be4web, 2016-10-06:
Hi,

can you double check that the file `./vcompile/build_rtl_sim.csh` in the `vsim/` directory really exists? If you do not have it, just pull it from github.

My other idea why that might fail is that you don't have the C Shell installed. You might want to check that `tcsh` is installed. Did you create the build directory in the sw folder? Otherwise you need to adapt some paths of the `cmake_configure` script.

Florian

samiasd, 2016-10-06:
thank you for reply,,,,
I'm sorry , I forgot to update the case, i fixed this problem and now I got a differenet error when I tried to execute the Hello World as followed

/pulpino/build:$make helloworld.vsim
Scanning dependencies of target string
[ 20%] Building C object libs/string_lib/CMakeFiles/string.dir/src/qprintf.c.o
Linking C static library libstring.a
[ 20%] Built target string
Scanning dependencies of target Arduino_core
[ 20%] Building C object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/wiring_digital.c.o
In file included from /home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/inc/Arduino.h:34:0,
                 from /home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/inc/wiring_private.h:32,
                 from /home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/src/wiring_digital.c:28:
/home/salamin/pulpino/sw/libs/sys_lib/inc/int.h: In function ‘int_disable’:
/home/salamin/pulpino/sw/libs/sys_lib/inc/int.h:53:3: warning: implicit declaration of function ‘mtspr’ [-Wimplicit-function-declaration]
   mtspr(SPR_SR, mfspr(SPR_SR) & (~SPR_SR_IEE));
   ^
/home/salamin/pulpino/sw/libs/sys_lib/inc/int.h:53:3: warning: implicit declaration of function ‘mfspr’ [-Wimplicit-function-declaration]
In file included from /home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/inc/wiring_private.h:35:0,
                 from /home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/src/wiring_digital.c:28:
/home/salamin/pulpino/sw/libs/sys_lib/inc/utils.h: At top level:
/home/salamin/pulpino/sw/libs/sys_lib/inc/utils.h:72:20: warning: conflicting types for ‘mtspr’ [enabled by default]
 static inline void mtspr(unsigned long spr, unsigned long value)
                    ^
/home/salamin/pulpino/sw/libs/sys_lib/inc/utils.h:72:20: error: static declaration of ‘mtspr’ follows non-static declaration
In file included from /home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/inc/Arduino.h:34:0,
                 from /home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/inc/wiring_private.h:32,
                 from /home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/src/wiring_digital.c:28:
/home/salamin/pulpino/sw/libs/sys_lib/inc/int.h:53:3: note: previous implicit declaration of ‘mtspr’ was here
   mtspr(SPR_SR, mfspr(SPR_SR) & (~SPR_SR_IEE));
   ^
In file included from /home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/inc/wiring_private.h:35:0,
                 from /home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/src/wiring_digital.c:28:
/home/salamin/pulpino/sw/libs/sys_lib/inc/utils.h:78:29: error: conflicting types for ‘mfspr’
 static inline unsigned long mfspr(unsigned long spr)
                             ^
In file included from /home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/inc/Arduino.h:34:0,
                 from /home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/inc/wiring_private.h:32,
                 from /home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/src/wiring_digital.c:28:
/home/salamin/pulpino/sw/libs/sys_lib/inc/int.h:53:17: note: previous implicit declaration of ‘mfspr’ was here
   mtspr(SPR_SR, mfspr(SPR_SR) & (~SPR_SR_IEE));
                 ^
/home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/src/wiring_digital.c: In function ‘pinMode’:
/home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/src/wiring_digital.c:43:3: warning: implicit declaration of function ‘csrr’ [-Wimplicit-function-declaration]
   csrr(mstatus, oldMstatus);
   ^
/home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/src/wiring_digital.c:43:8: error: ‘mstatus’ undeclared (first use in this function)
   csrr(mstatus, oldMstatus);
        ^
/home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/src/wiring_digital.c:43:8: note: each undeclared identifier is reported only once for each function it appears in
/home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/src/wiring_digital.c:46:3: warning: implicit declaration of function ‘csrw’ [-Wimplicit-function-declaration]
   csrw(mstatus, oldMstatus);
   ^
/home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/src/wiring_digital.c: In function ‘digitalWrite’:
/home/salamin/pulpino/sw/libs/Arduino_lib/core_libs/src/wiring_digital.c:84:7: error: ‘mstatus’ undeclared (first use in this function)
  csrr(mstatus, oldMstatus);
       ^
make[3]: *** [libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/wiring_digital.c.o] Error 1
make[2]: *** [libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/all] Error 2
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
make: *** [helloworld.vsim] Error 2

could you advise ??

closed 2016-10-06 by samiasd