pulp-platform/pulpino #38: Question about spiload error.

bobyan1979, 2016-10-18:
Hi,
I use the pulpino-zedboard-images you provided, partition my sd card as you advised. After linux is on, I run the command:

> ./spiload helloworld.stim.txt
 
But, on the screen, the following error is showing:

> Read check failed at idx xxx: Expected xx, got 00

It seams that nothing is writen into the Inst. RAM & data RAM, but the fpga seams is configured OK, because the Blue led is on after the system is brought up.

Any suggestions for me? thx.

gautschimi, 2016-10-19:
hi

can you check if there is reasonable data in the file helloworld.stim.txt?

did you connect over uart or ssh to the board?

best,

michael

bobyan1979, 2016-10-19:
Thanks for your reply.
I have checked the file helloworld.stim.txt, it's a file like

```
00000000_xxxxxxxx
00000004_xxxxxxxx
... ...
``` 
(xxxxxxxx is normal hex data, and the address includes inst. ram address and data ram address) 

I think the file is reasonable because I have run it ok in Modelsim, then copied the file spi_stim.txt and renamed it, everything seems OK.

Yes, I have connect the uart to the board. 

I found that when I use the boot.bin in the sd card attached of the zedboard, at the time of manual uboot, I print the cmd of "fpga info 0",  the 'cookie' field is "0x20(32)", but when I use the boot.bin in the tar.gz the project provided, at the time of manual uboot, I also print the cmd of "fpga info 0",  the 'cookie' filed is "0x0", Is the reason that we use the different versions of zedboard? Will it lead to a unworkable result?

I have no idea to check whether the bitstream in the boot.bin is download correctly and workable into the zedboard except the LED12 is on, any suggestion about it?

Thanks.

gautschimi, 2016-10-19:
Ok but there must be something wrong if the values in your 
helloworld.stim.txt are 'x'. that means they are not defined. is 
everything 'x'? are you sure the file has been generated? (with e.g. 
"make helloworld.stim.txt")

the spiload checks if everything has been written to the fpga. but 'x' 
is not a real value so when you read back what you have written it tries 
to compare the read value '0' with what you tried to write 'x' and this 
check obviously fails.

I would check if the helloworld has been properly generated. (check 
timestamp of the file and its content)
in my case the the helloworld.stim.txt looks like this:

00000000_00000013
00000004_00000013
00000008_00000013
....

you can also check this file:
...../sw/build/apps/helloworld/slm_files/spi_stim.txt

best,
michael

bobyan1979, 2016-10-19:
My stim file is just like yours.

Sorry, I use the character 'x' to represent a hex digit, which makes you mistaken my points.

gautschimi, 2016-10-19:
ahh ok now it makes more sense.

ok so let me summarize. you downloaded the bitstring from the repository 
and tried to launch helloworld by using the spiload. but you are using a 
different version of the zedboard? what version?

Hm I would try to specify a timeout when launching spiload or a 
different program. if it doesnt work you could try to build the 
bitstring with the flow described in the readme of the fpga folder

bobyan1979, 2016-10-19:
I remember it's "Xilinx Zynq-7000 AP SoC XC7Z020-CLG484" 

Does the zedboards made in different years can work normly with the the bitstring from the repository?
Thank you very much.

[image]

gautschimi, 2016-10-19:
> 2825494976

I think it should work yes. but I'm not 100% sure

bobyan1979, 2016-10-20:
I have check the board part information of my zedboard, the Digilent engineer tells me that I should use this board and part info.

```
create_project test_project /home/bobyan/test/test_project  -part xc7z020clg484-1
set_property board_part digilentinc.com:zedboard:part0:1.0 [current_project]
```

This is different from the board info in the tcl file of the PULP project. Which is like:
```
part: xc7z020clg484-1
board: "em.avnet.com:zynq:zed:c"
```
Should the bitstream file be compatible?
If  not, Maybe I should rebuild the bitstream file by using the board infomation provided by Digilent engineer.

gautschimi, 2016-10-20:
I don't know if the two boards are compatible. I would try to rebuild 
the bitstream for your board yes

bobyan1979, 2016-10-22:
Hi,

When I build ‘xilinx_clock_manager’, the following warning is occurred:

> # set_property -dict [list CONFIG.INTERFACE_SELECTION {Enable_AXI} CONFIG.USE_DYN_RECONFIG {true} CONFIG.PRIM_IN_FREQ {50.000} CONFIG.PRIMARY_PORT {clk50_i} CONFIG.CLK_OUT1_PORT {clk_o} CONFIG.CLKOUT1_REQUESTED_OUT_FREQ {5.000} CONFIG.LOCKED_PORT {rst_no} CONFIG.RESET_TYPE {ACTIVE_LOW} CONFIG.RESET_PORT {rst_ni}] [get_ips xilinx_clock_manager]
> WARNING: [IP_Flow 19-3374] An attempt to modify the value of disabled parameter 'RESET_TYPE' from 'ACTIVE_HIGH' to 'ACTIVE_LOW' has been ignored for IP 'xilinx_clock_manager'

Will it lead a wrong reset behavior?
Thanks

bobyan1979, 2016-10-24:
I have rebuild the project with the official board files by using vivado 2015.2. After some minor changes, it works in my zedboard.
Thank you @gautschimi .
Now I can close this Issue.

closed 2016-10-24 by bobyan1979