pulp-platform/pulpino #39: core registers debug

dcoriat, 2016-10-24:
Hi all,

I've been taking a look at the way the debug is done. I have a hard time comprehending how does one reads the values of the registers inside the core when the debug port of the core instantiation is left unconnected (see core_region.sv).

Am I missing something?

(Sorry if this is not the correct place to post this, please redirect me if needed)

Best regards,
David

be4web, 2016-11-07:
Hi,

the debug port you are mentioning is legacy code and in use anymore, hence it is not connected. The debug unit allowed us to differentiate between an AXI access and a core access. To simplify things we do not make this distinction anymore and everything is accessible through the AXI bus. 

Best,

Florian

dcoriat, 2016-11-08:
Hello Florian,

Thanks for your response. I still do not understand how is the debug port of riscv_core not used as it is the only port capable of responding to a request rather than initiate one (ie. slave against master).

Now let's say the debug unit sends a request on the AXI bus, requesting the value of the program counter for example. Can you help me understand which AXI slave answers?

Best regards,
David

gautschimi, 2016-11-08:
hello

all debug registers are memory mapped in pulpino. meaning that if you 
want to access the program counter for example you can access it over 
the debug bus which is connected to the apb peripherals.

the address range is this:
// MASTER PORT TO DEBUG
`define DEBUG_START_ADDR      32'h1A11_0000
`define DEBUG_END_ADDR        32'h1A11_7FFF

now since all apb peripherals are connected to the axi bus, and that 
jtag and spi have an axi-port you can actually access all debug 
registers over spi or jtag.
hope this helps, let me know if you have further questions

best,
michael

dcoriat, 2016-11-09:
Oh wow, totally my bad. I see it now. Some initial assumptions made me think wrong.

Sorry and thanks for taking the time to answer my questions.
I'll close the issue now.

All the best,
David

closed 2016-11-09 by dcoriat