pulp-platform/pulpino #41: make vcompile error

jpavieira, 2016-11-28:
When I run `make vcompile` in sw/build directory I get the following output errors:

```
$make vcompile

--> Compiling PULPino Platform... 

--> Compiling PULP IPs libraries... 
--> Compiling apb_gpio... 
Model Technology ModelSim ALTERA vmap 10.4d Lib Mapping Utility 2015.12 Dec 30 2015
vmap apb_gpio_lib /mnt/sda5/Dropbox/EE/Mestrado/Tese/pulpino/vsim/modelsim_libs/apb_gpio_lib 
Modifying modelsim.ini
Compiling component:  apb_gpio 

iet -- not found.
-sv -- not found.
-work -- not found.

/mnt/sda5/Dropbox/EE/Mestrado/Tese/pulpino/vsim/modelsim_libs/apb_gpio_lib:
/mnt/sda5/Dropbox/EE/Mestrado/Tese/pulpino/vsim/..//ips/apb/apb_gpio/apb_gpio.sv -- not found.

CMakeFiles/vcompile.dir/build.make:57: recipe for target 'CMakeFiles/vcompile' failed
make[3]: *** [CMakeFiles/vcompile] Error 1
CMakeFiles/Makefile2:168: recipe for target 'CMakeFiles/vcompile.dir/all' failed
make[2]: *** [CMakeFiles/vcompile.dir/all] Error 2
CMakeFiles/Makefile2:175: recipe for target 'CMakeFiles/vcompile.dir/rule' failed
make[1]: *** [CMakeFiles/vcompile.dir/rule] Error 2
Makefile:168: recipe for target 'vcompile' failed
make: *** [vcompile] Error 2
```
I don't understand what is happening, this is the ModelSim altera starter edition 10.4d 32bit.
I've already removed all -64 flags from the files indicated in other issues.

Why is it not generating the work folder?
I've already checked and the file apb_gpio.sv is in that location, there is no apparent reason to "not found".

Thank you!

gautschimi, 2016-11-28:
can you try to start the script directly from the modelsim folder:

in:
xxx/pulpino/vsim/

type:
"source vcompile/build_rtl_sim.csh"

in that case you don't have any cmake in between, but I'm not convinced 
it will solve your issues.

jpavieira, 2016-11-28:
Result:

```
$ source vcompile/build_rtl_sim.csh 
bash: vcompile/build_rtl_sim.csh: line 39: syntax error: unexpected end of file
```
I also tried to run it like `./vcompile/build_rtl_sim.csh` and the result was the same as before:
```
--> Compiling PULPino Platform... 

--> Compiling PULP IPs libraries... 
--> Compiling apb_gpio... 
Model Technology ModelSim ALTERA vmap 10.4d Lib Mapping Utility 2015.12 Dec 30 2015
vmap apb_gpio_lib /mnt/sda5/Dropbox/EE/Mestrado/Tese/pulpino/vsim/modelsim_libs/apb_gpio_lib 
Modifying modelsim.ini
Compiling component:  apb_gpio 

iet -- not found.
-sv -- not found.
-work -- not found.

/mnt/sda5/Dropbox/EE/Mestrado/Tese/pulpino/vsim/modelsim_libs/apb_gpio_lib:
/mnt/sda5/Dropbox/EE/Mestrado/Tese/pulpino/vsim/..//ips/apb/apb_gpio/apb_gpio.sv -- not found.

```

gautschimi, 2016-11-28:
something is wrong with:
a) your modelsim installation or
b) your shell and set paths

maybe just try to build some files by hand by looking at some of the 
vcompile scripts. e.g. vsim/vcompile/ips/vcompile_riscv
within vsim folder:

1) try to make a library: "vlib riscv_lib"
2) now compile a system verilog file into this library: "vlog -sv -work 
riscv_lib $IP_PATH/riscv/includes/riscv_defines.sv"
3) try the same with vhdl "vcom -library riscv_lib $someVHDLfile"

if that doesnt work, well then your modelsim is not set up properly

jpavieira, 2016-11-28:
I found out that I have a vlog command as native in my /usr/bin directory, when the script executes vlog it uses this native command and not the modelsim one. 
How can I override the native command?
I've tried an alias in .profile but with no effect...

closed 2016-12-02 by jpavieira