pulp-platform/pulpino #42: make helloworld.vsim error

micheleportolan, 2016-11-30:
I finally got vcompile to work with no errors, but now I am stuck at the last step : 

make helloworld.vsim
[  0%] Building C object libs/bench_lib/CMakeFiles/bench.dir/src/bench.c.o
/calcul/pulpino/pulpino/sw/libs/sys_lib/inc/utils.h: Assembler messages:
/calcul/pulpino/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/calcul/pulpino/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/calcul/pulpino/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/calcul/pulpino/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/calcul/pulpino/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/calcul/pulpino/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/calcul/pulpino/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/calcul/pulpino/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/calcul/pulpino/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/calcul/pulpino/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/calcul/pulpino/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/calcul/pulpino/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/calcul/pulpino/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
make[3]: *** [libs/bench_lib/CMakeFiles/bench.dir/src/bench.c.o] Error 1
make[2]: *** [libs/bench_lib/CMakeFiles/bench.dir/all] Error 2
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
make: *** [helloworld.vsim] Error 2

What could the error be? I use gcc 5.4.0

Thanks,

Michele

gautschimi, 2016-12-01:
the instructions l.mfspr xxx are no valid risc-v instructions, it seems 
that you are compiling the OpenRISC version..

I guess you don't do that on purpose. you probably used the wrong script 
to setup the build folder.

this is the correct one:
cmake_configure.riscv.gcc.sh

micheleportolan, 2016-12-01:
I noticed this morning that my ./build directory was in the pulpino root. I re-run the flow in the ./sw directory (so ./sw/build) and the compilation ends correctly: cmake was probably making some erroneous links due to the different relative path. 
Thanks,

Michele

closed 2016-12-01 by micheleportolan