pulp-platform/pulpino #43: fpga build fails; missing tcl files.

proteus-cpi, 2016-12-01:
Apparently the latest version (7ad47f3761db74a8f681d952e125664fd5abe71e) build fails due to missing tcl files.

Log:
##################################################################################

proteus-cpi@ACD-K20:/Workspace/Workspace-PULPINO/pulpino$ cd fpga/
proteus-cpi@ACD-K20:/Workspace/Workspace-PULPINO/pulpino/fpga$ make 
make -C pulpino clean all
make[1]: Entering directory '/Workspace/Workspace-PULPINO/pulpino/fpga/pulpino'
rm -rf pulpino.xpr
rm -rf pulpino.data
rm -rf pulpino.runs
rm -rf pulpino.srcs
rm -rf pulpino.sim
rm -rf pulpino.cache
rm -rf pulpino.ioplanning
rm -rf pulpino.hw
rm -rf component.xml
rm -rf vivado*
rm -rf .Xil
rm -rf xgui
rm -rf pulpino.edf
rm -rf pulpino_stub.v
rm -rf pulpino_funcsim.v
vivado -mode batch -source tcl/run.tcl

****** Vivado v2015.2.1 (64-bit)
  **** SW Build 1302555 on Wed Aug  5 13:06:02 MDT 2015
  **** IP Build 1291990 on Mon Jul 27 03:18:52 MDT 2015
    ** Copyright 1986-2015 Xilinx, Inc. All Rights Reserved.

source tcl/run.tcl
# if { ![info exists ::env(BOARD) ]} {
#   set ::env(BOARD) "zedboard"
# }
# if { ![info exists ::env(XILINX_PART)] } {
#   set ::env(XILINX_PART) "xc7z020clg484-1"
# }
# if { ![info exists ::env(XILINX_BOARD)] } {
#   set ::env(XILINX_BOARD) "em.avnet.com:zynq:zed:c"
# }
# create_project pulpino . -part $::env(XILINX_PART)
# set_property board $::env(XILINX_BOARD) [current_project]
WARNING: [Common 17-599] Property 'board' is deprecated for object type 'project'. Use 'board_part'.
INFO: [Board 49-70] Updating board vlnv to compatible board_part vlnv
# source tcl/ips_inc_dirs.tcl
couldn't read file "tcl/ips_inc_dirs.tcl": no such file or directory
    while executing
"source tcl/ips_inc_dirs.tcl"
    (file "tcl/run.tcl" line 18)
INFO: [Common 17-206] Exiting Vivado at Thu Dec  1 15:59:18 2016...
Makefile:4: recipe for target 'all' failed
make[1]: *** [all] Error 1
make[1]: Leaving directory '/Workspace/Workspace-PULPINO/pulpino/fpga/pulpino'
Makefile:35: recipe for target 'pulpino/pulpino.edf' failed
make: *** [pulpino/pulpino.edf] Error 2
proteus-cpi@ACD-K20:/Workspace/Workspace-PULPINO/pulpino/fpga

####################################################################

gautschimi, 2016-12-01:
In the readme you will find that you first have to download all ips. 
this is done by executing the python script "update_ips.py"

do this and try again please

jpavieira, 2016-12-02:
Same problem here. The update_ips.py didn't solve the problem...

gautschimi, 2016-12-04:
if it completes without error it will generate the scripts. in any case, 
try to run "generate_scripts" then

closed 2017-02-09 by gautschimi

suhasskrishanmurthy, 2019-06-19:
I tried both the things that you mentioned, but when i clone the repository ips_lists.txt file is not present and after executing the python script "update_ips.py", still the ips_inc_dirs.tcl file is missing. How can i resolve this issue?

SubhamRath, 2021-03-02:
Facing the same issue, Can you please help in resolving that?