pulp-platform/pulpino #47: Spiload and GLIBC version issue

marci89, 2016-12-13:
Hello,
I'm trying to run a custom program on the pulpino platform. I've accomplished all the steps as the readme file shows (the one within the FPGA folder). I have a running version of Linux (xilinx version) on a zybo development board and I can run commands correctly with both UART and ssh.
When I try to run spiload in order to run my custom program I get this error:

./spiload: /lib/libc.so.6: version `GLIBC_2.17' not found (required by ./spiload)

I compiled spiload, as well as the xilinx linux, with arm-xilinx-linux-gnueabi-gcc 4.9.1.

Anyone that got this error could help me?

Thank you very much

marci89, 2016-12-14:
[image]

gautschimi, 2016-12-14:
> img_20161213_153053

Does that only happen when you run your custom application or for all 
applications?

can you check if you have glibc_2.17 on the arm platform?

marci89, 2016-12-14:
hi gautschimi,

it happens when I run spiload, with whichever application.

when I run /lib/libc.so.6 through the arm terminal, I get the message in the screenshot below

[image]

so the version seems to be GNU C Library version 2.11.1

As I said, I compiled the xilinx linux, as well as spiload, with the xilinx toolchain (arm-xilinx-linux-gnueabi-gcc 4.9.1)

It seems that the glib used within the arm platform has been pre-compiled and the one that I use for compiling spiload is newer. But I've no idea of how I can upgrade the glibc in the arm platform.

closed 2017-01-08 by marci89