pulp-platform/pulpino #50: synthesis fails on vivado 2015.4 giving the following errors

abdallah93as, 2017-01-11:
Hello, i followed your instructions and run both scripts: "update-ips.py" and "generate-scripts.py".

I opened a new project in Vivado and added sources by choosing add directories, and chose "pulpemu-top" as top module. Running the synthesis gives me the foloowing errors, pls help.

*** Running vivado
    with args -log pulpemu_top.vds -m64 -mode batch -messageDb vivado.pb -notrace -source pulpemu_top.tcl

****** Vivado v2015.4 (64-bit)
  **** SW Build 1412921 on Wed Nov 18 09:44:32 MST 2015
  **** IP Build 1412160 on Tue Nov 17 13:47:24 MST 2015
    ** Copyright 1986-2015 Xilinx, Inc. All Rights Reserved.

source pulpemu_top.tcl -notrace
Command: synth_design -top pulpemu_top -part xc7z045ffg900-2
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xc7z045'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7z045'
INFO: Launching helper process for spawning children vivado processes
INFO: Helper process launched with PID 5369 
ERROR: [Synth 8-1766] cannot open include file defines_event_unit.sv [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:11]
ERROR: [Synth 8-2841] use of undefined macro REGS_SLEEP_MAX_IDX [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:38]
ERROR: [Synth 8-2715] syntax error near ] [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:38]
ERROR: [Synth 8-2841] use of undefined macro REG_SLEEP_CTRL [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:65]
ERROR: [Synth 8-2841] use of undefined macro SLEEP_ENABLE [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:65]
ERROR: [Synth 8-2715] syntax error near ] [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:65]
ERROR: [Synth 8-2715] syntax error near ] [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:110]
ERROR: [Synth 8-2841] use of undefined macro REG_SLEEP_STATUS [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:163]
ERROR: [Synth 8-2841] use of undefined macro SLEEP_STATUS [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:163]
ERROR: [Synth 8-2715] syntax error near ] [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:163]
ERROR: [Synth 8-2715] syntax error near ] [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:167]
ERROR: [Synth 8-2715] syntax error near : [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:174]
ERROR: [Synth 8-2715] syntax error near ] [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:175]
ERROR: [Synth 8-2715] syntax error near : [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:193]
ERROR: [Synth 8-2715] syntax error near ] [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:194]
ERROR: [Synth 8-2715] syntax error near : [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:196]
ERROR: [Synth 8-2715] syntax error near ] [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:197]
ERROR: [Synth 8-1031] regs_q is not declared [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:160]
ERROR: [Synth 8-1875] default case should appear only once [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:197]
ERROR: [Synth 8-1875] default case should appear only once [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:200]
ERROR: [Synth 8-1031] regs_q is not declared [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:213]
ERROR: [Synth 8-1031] regs_q is not declared [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:219]
INFO: [Synth 8-2350] module sleep_unit ignored due to previous errors [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv:13]
Failed to read verilog '/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv'
INFO: [Common 17-83] Releasing license: Synthesis
3 Infos, 0 Warnings, 0 Critical Warnings and 23 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Vivado Synthesis failed
INFO: [Common 17-206] Exiting Vivado at Wed Jan 11 11:36:06 2017...

gautschimi, 2017-01-11:
well I don't know but check if the paths are correct. it seems like it 
cannot find the file"
defines_event_unit.sv which was included by 
/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/apb/apb_event_unit/sleep_unit.sv

I'd suggest to use our flow to generate the vivado project file, and 
then open this project and do your modifications.

abdallah93as, 2017-01-11:
Hello, and thanks for your reply. So the errors were due to having 2 files not set as "verilog headers".
So i set "defines_event_unit.sv" and "riscv_config.sv" as verilog headers, and as global include files, and luckily the errors were gone. But still, some new errors emerged, below is the log file.

Also i have a question, I did not understand what you meant when you said: 
" I'd suggest to use our flow to generate the Vivado project file, and then open this project and do your modifications."

I did not find such a thing in the readme file that is included.

*** Running vivado
    with args -log pulpemu_top.vds -m64 -mode batch -messageDb vivado.pb -notrace -source pulpemu_top.tcl

****** Vivado v2015.4 (64-bit)
  **** SW Build 1412921 on Wed Nov 18 09:44:32 MST 2015
  **** IP Build 1412160 on Tue Nov 17 13:47:24 MST 2015
    ** Copyright 1986-2015 Xilinx, Inc. All Rights Reserved.

source pulpemu_top.tcl -notrace
Command: synth_design -top pulpemu_top -part xc7z045ffg900-2
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xc7z045'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7z045'
INFO: Launching helper process for spawning children vivado processes
INFO: Helper process launched with PID 8509 
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:169]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:170]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:171]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:178]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:179]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:180]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:187]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:188]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:195]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:196]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:197]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:198]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:205]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:206]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:213]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:214]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:221]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:222]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv:229]
Failed to read verilog '/root/Vivado/Pulpin3/Pulpin3.srcs/sources_1/imports/pulpino/ips/riscv/riscv_tracer.sv'
INFO: [Common 17-83] Releasing license: Synthesis
2 Infos, 0 Warnings, 0 Critical Warnings and 20 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Vivado Synthesis failed
INFO: [Common 17-206] Exiting Vivado at Wed Jan 11 13:40:16 2017...

gautschimi, 2017-01-11:
there are makefiles in the fpga folder which create all this.

abdallah93as, 2017-01-11:
ok i ran the Makefile using "make all", it gave the following error at the end. Should I also download the 2015.1 or is it ok to still test it on 2015.4

## if { [string first $scripts_vivado_version $current_vivado_version] == -1 } {
##    puts ""
##    puts "ERROR: This script was generated using Vivado <$scripts_vivado_version> and is being run in <$current_vivado_version> of Vivado. Please run the script in Vivado <$scripts_vivado_version> then open the design in Vivado <$current_vivado_version>. Upgrade the design by running \"Tools => Report => Report IP Status...\", then run write_bd_tcl to create an updated script."
## 
##    return 1
## }

ERROR: This script was generated using Vivado <2015.1> and is being run in <2015.4> of Vivado. Please run the script in Vivado <2015.1> then open the design in Vivado <2015.4>. Upgrade the design by running "Tools => Report => Report IP Status...", then run write_bd_tcl to create an updated script.
# validate_bd_design
ERROR: [BD 5-104] A block design must be open to run this command. Please create/open a block design.
ERROR: [Common 17-39] 'validate_bd_design' failed due to earlier errors.

    while executing
"validate_bd_design"
    (file "tcl/run.tcl" line 40)
INFO: [Common 17-206] Exiting Vivado at Wed Jan 11 15:48:16 2017...
Makefile:6: recipe for target 'synth' failed
make[1]: *** [synth] Error 1
make[1]: Leaving directory '/root/Desktop/Git/pulpino/fpga/pulpemu'
Makefile:42: recipe for target 'pulpemu/pulpemu_top.bit' failed
make: *** [pulpemu/pulpemu_top.bit] Error 2

abdallah93as, 2017-01-12:
ok it's resolved i changed the value from 2015.1 to 2015.4 in the tcl file that gave me the error. The synthesis flow is working fine now. There is an error i believe in the final bitstream. But that is not important for now, when the time comes I'll ask about it

closed 2017-01-15 by abdallah93as