pulp-platform/pulpino #55: ./update-ips.py need password

rdapa, 2017-02-09:
Hi  administrator
**when i** 
./update-ips.py git@iis-git.ee.ethz.ch   || exit 1

**show these**
Using remote git server git@iis-git.ee.ethz.ch, remote is git@iis-git.ee.ethz.ch:pulp-project
Initialized empty Git repository in /home/icer/projects/pulp_my/pulpino-master/ipstools/.git/
git@iis-git.ee.ethz.ch's password: 

what is the password ?

rdapa, 2017-02-09:
like these 
[icer@192 pulpino-master]$ ./update-ips.py git@iis-git.ee.ethz.ch
Using remote git server git@iis-git.ee.ethz.ch, remote is git@iis-git.ee.ethz.ch:pulp-project
Initialized empty Git repository in /home/icer/projects/pulp_my/pulpino-master/ipstools/.git/
The authenticity of host 'iis-git.ee.ethz.ch (129.132.2.24)' can't be established.
RSA key fingerprint is XXXXXXXX
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'iis-git.ee.ethz.ch,129.132.2.24' (RSA) to the list of known hosts.
git@iis-git.ee.ethz.ch's password:

gautschimi, 2017-02-09:
You should use github as remote server: git@github.com

infact, why do you use update_ips.py like this? Is that written somewhere?

rdapa, 2017-02-09:
its written in /ci/setup.csh.    

I use these , it does not work either.
[icer@192 pulpino-master]$ ./update-ips.py git@github.com
Using remote git server git@github.com, remote is git@github.com:pulp-project
Initialized empty Git repository in /home/icer/projects/pulp_my/pulpino-master/ipstools/.git/
Permission denied (publickey).
fatal: The remote end hung up unexpectedly
Initialized empty Git repository in /home/icer/projects/pulp_my/pulpino-master/ipstools/.git/
Permission denied (publickey).
fatal: The remote end hung up unexpectedly
Traceback (most recent call last):
  File "./update-ips.py", line 86, in <module>
    import ipstools
ImportError: No module named ipstools

gautschimi, 2017-02-09:
ok, that is our continuous integration framework which is located on our own server. You can not use it.

instead you should follow the setup which is described in the README.md of the pulpino folder.
update_ips.py should connect to github and use your own github password (if you have not set the ssh key)

have you downloaded the zip? or did you clone the sources from github?

closed 2017-02-23 by gautschimi