pulp-platform/pulpino #56: Zedboard FPGA verification is not working

satya123bora, 2017-02-16:
I have generated and copied BOOT.BIN, uImage and devicetree.dtb files to the first partition of SD card. The rootfs.tar is also extracted inside the second partition of SD card. After that if we try to boot from SD card it sows the following error:

U-Boot 2015.01-dirty (Feb 13 2017 - 14:53:14)                                   
                                                                                
Board:  Xilinx Zynq                                                             
I2C:   ready                                                                    
DRAM:  ECC disabled 512 MiB                                                     
MMC:   zynq_sdhci: 0                                                            
SF: Detected S25FL256S_64K with page size 256 Bytes, erase size 64 KiB, total 3B
*** Warning - bad CRC, using default environment                                
                                                                                
In:    serial                                                                   
Out:   serial                                                                   
Err:   serial                                                                   
Board:  Xilinx Zynq                                                             
Net:   Gem.e000b000                                                             
Error: Gem.e000b000 address not set.

Hit any key to stop autoboot:  3

If I allow atutoboot, it gets stuck at the following point:

can: netlink gateway (rev 20130117) max_hops=1                                  
Registering SWP/SWPB emulation handler                                          
mmc0: new high speed SDHC card at address 0007                                  
mmcblk0: mmc0:0007 DDINC 3.70 GiB                                               
 mmcblk0: p1 p2                                                                 
drivers/rtc/hctosys.c: unable to open rtc device (rtc0)                         
Sending DHCP and RARP requests ....                                             
xemacps e000b000.ethernet: Set clk to 25000000 Hz                               
xemacps e000b000.ethernet: link up (100/FULL)                                   
.. timed out!                                                                   
IP-Config: Reopening network devices...                                         
Sending DHCP and RARP requests ..                                               
xemacps e000b000.ethernet: Set clk to 25000000 Hz                               
xemacps e000b000.ethernet: link up (100/FULL)                                   
...

The ethernet cable is connected to the host PC. What could be the possible reason of error? Please have a look to the matter.

gautschimi, 2017-02-17:
hello

Have you prepared the SD card as described in [1]? the boot partition 
should have a fat32 filesystem and the root partition an ext4.

The BOOT.BIN, uImage and devicetree.dtb should go to the first partition 
and the rootfs.tar to the second.

[1] http://www.wiki.xilinx.com/Prepare+Boot+Medium

regarding the Ethernet connection. it seems that the clock with 25MHz is 
wrong. It should be 125MHz. It is probably just using a default 
configuration. I guess it's a problem in the device tree. I'd regenerate 
everything and try again

gautschimi, 2017-02-17:
I just consulted with a colleague and he think it might be because you 
have another configuration in the flash.
to fix this and load the variables from the sd card you have to clear 
them before the autoboot.

you can do it with the following commands:
"env default -a"
"saveenv"

then you have to reboot. (you only have to do this once)

best,
Michael

satya123bora, 2017-02-17:
I did 

"env default -a"
"saveenv"

Now it is going to the further steps with the following messages:

drivers/rtc/hctosys.c: unable to open rtc device (rtc0)
Sending DHCP and RARP requests ...... timed out!
IP-Config: Reopening network devices...
Sending DHCP and RARP requests ...... timed out!
IP-Config: Auto-configuration of network failed
ALSA device list:
  No soundcards found.
EXT3-fs (mmcblk0p2): error: couldn't mount because of unsupported optional features (240)
EXT2-fs (mmcblk0p2): error: couldn't mount because of unsupported optional features (244)
EXT4-fs (mmcblk0p2): recovery complete
EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
VFS: Mounted root (ext4 filesystem) on device 179:2.
devtmpfs: mounted
Freeing unused kernel memory: 216K (40634000 - 4066a000)
Starting logging: OK
Initializing random number generator... random: dd urandom read with 6 bits of entropy available
done.
Starting network...
ip: RTNETLINK answers: Operation not permitted
Starting sshd: key_load_public: invalid format
Could not load host key: /etc/ssh/ssh_host_rsa_key
key_load_public: invalid format
Could not load host key: /etc/ssh/ssh_host_dsa_key
key_load_public: invalid format
Could not load host key: /etc/ssh/ssh_host_ecdsa_key
key_load_public: invalid format
Could not load host key: /etc/ssh/ssh_host_ed25519_key
Disabling protocol version 2. Could not load host key
sshd: no hostkeys available -- exiting.
OK
can't open /dev/console: Permission denied
can't open /dev/console: Permission denied
can't open /dev/console: Permission denied

Do I need to connect the zedbozrd with local network using ethernet? If we leave the ethernet port open then also it should work, am I right?

satya123bora, 2017-02-23:
The core is now working. The linux is working properly. I did it from the beginning and now it is working fine.
Thank you for help and valuable time.

Regards,
Satya

closed 2017-02-23 by satya123bora

soyedtuhin, 2018-11-07:
I am having the same issue 

U-Boot 2015.01-dirty (Nov 02 2018 - 14:31:41)

Board:  Xilinx Zynq
I2C:   ready
DRAM:  ECC disabled 512 MiB
MMC:   zynq_sdhci: 0
SF: Detected S25FL256S_64K with page size 256 Bytes, erase size 64 KiB, total 32 MiB
In:    serial
Out:   serial
Err:   serial
Board:  Xilinx Zynq
Net:   Gem.e000b000
Error: Gem.e000b000 address not set.

If I allow atutoboot, i get the following point time out:
mmcblk0: mmc0:aaaa SS16G 14.8 GiB
Sending DHCP and RARP requests .
 mmcblk0: p1 p2
..... timed out!
IP-Config: Reopening network devices...
Sending DHCP and RARP requests ...... timed out!
IP-Config: Auto-configuration of network failed

then gets stuck at the following point:
Welcome to Buildroot
buildroot login:

It does not go to the expected "zynq>"  

boot process file is attached