pulp-platform/pulpino #58: Implement on Artix-7 FPGA

markmatthewanderson, 2017-02-18:
Hi,

I'm working on a project, and we're trying to put the PULPino on an Artix-7 FPGA part number xc7a100tftg256-2 so we can do side-channel analysis. We've gone through and changed the XILINX_PART variable in the .tcl files in the sub-directories under /pulpino/fpga/ips. We're getting the following error because we're working on an Artix FPGA and not the ZYNQ PSoC:

ERROR: [BD 5-216] VLNV <xilinx.com:ip:processing_system7:5.5> is not supported for the current part.

Do you have any tips for getting past this error and implementing the PULPino on our FPGA?

Thank you!

gautschimi, 2017-02-19:
Hello,

the reason is that you don't have a processing system on the artix7.

the fpga pulpino is designed to work with an arm core. well it's not a 
requirement, but the way it is set up it will not work without.

So if you want to port it to Artix 7 you should proceed a bit 
differently. You should make your own project, instantiate the toplevel 
of pulpino and then connect the pulpino ports to some pins, but not to 
the processing system.

best,

michael

closed 2017-04-25 by gautschimi

SubhamRath, 2021-03-03:
Hello, @markmatthewanderson can you please elaborate the steps which you followed to set the XILINX_PART and XILINX_BOARD in the tcl files?