pulp-platform/pulpino #60: error: unrecognized command line option '-m32'

farzadsamie, 2017-02-21:
I installed the official RISC-V toolchain from Berkeley for 32-bit machine (riscv32-unknown-elf-XXX are built).
However, when I run the cmake_configure.riscv.gcc.sh script I get this error:

error: unrecognized command line option '-m32'

The cross-compiler does not recognize this flag.

gautschimi, 2017-02-23:
hello

you should use our toolchain which you can find here:
https://github.com/pulp-platform/ri5cy_gnu_toolchain

(it is the official toolchain + our extensions). although it is not 
strictly required to use our toolchain, we recommend to use it. as you 
can see the official toolchain does not support all our flags, which 
means that you have to remove them in the CMakeList.txt files for each 
application.

best,

michael

farzadsamie, 2017-02-24:
@gautschimi 
Thanks, it solved the problem.

closed 2017-03-03 by gautschimi