pulp-platform/pulpino #63: Profile pulpino applications

valenaUsi, 2017-03-29:
Hello everyone,

I would like to profile some application for Pulpino but I have dome problems to do that. I have seen that it is possible to profile the application using kcachegrind but when I run make helloworld.kcg" I get the error "pulp-pc-analyze" does not exist. This is the line that throws the error pulp-pc-analyze --rtl --input=trace_core_00.log --binary=helloworld.elf.

So I would like to ask you if there is a way get the tool "pulp-pc-analyze" or how do you profile the applications with pulpino. I am thinking to get the all information from trace_core_00.log but i would prefer to use a tool that allows me to visualize the information

Thanks in advance!
Felipe

closed 2017-06-21 by gautschimi

stefanct, 2019-03-08:
May I ask why this was closed without comments? The KCachegrind functionality is still "advertised" in the readme but I could not find any source for this utility either and no other traces but the cmake references within the repository.

stefanct, 2019-03-08:
Likewise, `vsim/tcl_files/run_boot.tcl` is missing from the repository (needed for the `.vsim.boot` targets).