pulp-platform/pulpino #64: install modelsim

barry41409, 2017-03-30:
The pulpino require to install modelsim.
Which modelsim do I install ModelSim*-Intel FPGA(https://www.altera.com/products/design-software/model---simulation/modelsim-altera-software.html) or ModelSim PE Student Edition (https://www.mentor.com/company/higher_ed/modelsim-student-edition) ?
I check the version is over 10.2c.

If I install ModelSim PE Student Edition,I have problem about the path of bashrc.
I can't find the bin floder in the modelsim floder.

Installing ModelSim PE Student Edition is correct?

gautschimi, 2017-03-30:
pulpino works with any modelsim version. for some you might have to 
adjust a script. but the problem you are having is not related to pulpino.

so the student version is ok

closed 2017-04-25 by gautschimi

GioAgu17, 2017-05-15:
Sorry but I am having some troubles in figuring out what actually is the ModelSim required for PULPino.

I am using Linux and I went in the Altera website to download Quartus Pro Prime edition, and they recommend me to install also the device support. But it's 26GB of download!!

What software do I have to install in order to use PULPino from Linux (32 bit machine)?

Thank you so much for your attention

Giovanni

gautschimi, 2017-05-15:
Hello Giovanni

Pulpino does not strictly require a specific version of a simulator. We 
mainly used ModelSim/QuestaSim for simulation, and that is what is also 
supported by our automatic script generation tools for the IPs. We 
cannot currently provide support on other simulation environments, but 
the platform should be working on any simulation tool that is able to 
understand synthesizable SystemVerilog

GioAgu17, 2017-05-15:
Ok thank you again

Bests,
Giovanni