pulp-platform/pulpino #65: -m32 option is an unknown command for riscv32-unknown-elf-gcc

Aregaard, 2017-04-15:
Hi,

System: 
MacOS with running Parallels. 

Running Ubuntu 16.04 64-bit as guest OS.
Installed Modelsim 16.01 Starter Edition with 32-bit libs

Modelsim is working
compilers riscv32-unknown-elf-gcc and riscv64-unknown-elf-gcc works fine.
But "spike pk hello" is only for working for riscv64 and not riscv32. Complains about memory.

------

When I try to run the cmake_configure.riscv.gcc.sh the output gives an error:
"riscv32-unknown-elf-gcc: error: unrecognized command line option -m32"

If I remove the -m32 it builds OK.
"Configuration done"
"Generating done"

"Build files have been written to pulpino/sw/build"

But then if I run "make vcompile" in the build folder. I get:
(Last parts causing errors)

root@ubuntu:~/PULPino/pulpino/sw/build# make vcompile
Scanning dependencies of target vcompile

--> Compiling PULPino Platform... 

--> Compiling PULP IPs libraries... 
--> Compiling apb_gpio... 
Model Technology ModelSim - Intel FPGA Edition vmap 10.5b Lib Mapping Utility 2016.10 Oct  5 2016
vmap apb_gpio_lib /home/aregaard/PULPino/pulpino/vsim/modelsim_libs/apb_gpio_lib 
Modifying modelsim.ini
Compiling component:  apb_gpio 

--> apb_gpio compilation complete! 
--> Compiling axi_slice_dc... 
Model Technology ModelSim - Intel FPGA Edition vmap 10.5b Lib Mapping Utility 2016.10 Oct  5 2016
vmap axi_slice_dc_lib /home/aregaard/PULPino/pulpino/vsim/modelsim_libs/axi_slice_dc_lib 
Modifying modelsim.ini
Compiling component:  axi_slice_dc 

--> axi_slice_dc compilation complete! 
--> Compiling apb_event_unit... 
Model Technology ModelSim - Intel FPGA Edition vmap 10.5b Lib Mapping Utility 2016.10 Oct  5 2016
vmap apb_event_unit_lib /home/aregaard/PULPino/pulpino/vsim/modelsim_libs/apb_event_unit_lib 
Modifying modelsim.ini
Compiling component:  apb_event_unit 

--> apb_event_unit compilation complete! 
--> Compiling axi_node... 
Model Technology ModelSim - Intel FPGA Edition vmap 10.5b Lib Mapping Utility 2016.10 Oct  5 2016
vmap axi_node_lib /home/aregaard/PULPino/pulpino/vsim/modelsim_libs/axi_node_lib 
Modifying modelsim.ini
Compiling component:  axi_node 

--> axi_node compilation complete! 
--> Compiling riscv... 
Model Technology ModelSim - Intel FPGA Edition vmap 10.5b Lib Mapping Utility 2016.10 Oct  5 2016
vmap riscv_lib /home/aregaard/PULPino/pulpino/vsim/modelsim_libs/riscv_lib 
Modifying modelsim.ini
Compiling component:  riscv 

** Warning: /home/aregaard/PULPino/pulpino/vsim/..//ips/riscv/controller.sv(593): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Warning: /home/aregaard/PULPino/pulpino/vsim/..//ips/riscv/controller.sv(29): (vlog-2186) SystemVerilog testbench feature
[... 184 lines cut ...]
 Compiling for RISCV core 
** Warning: /home/aregaard/PULPino/pulpino/vsim/..//rtl/random_stalls.sv(62): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Warning: /home/aregaard/PULPino/pulpino/vsim/..//rtl/random_stalls.sv(106): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
--> PULPino compilation complete! 
--> Compiling work.tb... 
Compiling component:  work.tb 

Failed to open executable /home/aregaard/intelFPGA/16.1/modelsim_ase/linuxaloem/../linux_x86_64pe/vlog in execute mode needed for the option -64.

CMakeFiles/vcompile.dir/build.make:57: recipe for target 'CMakeFiles/vcompile' failed
make[3]: *** [CMakeFiles/vcompile] Error 1
CMakeFiles/Makefile2:168: recipe for target 'CMakeFiles/vcompile.dir/all' failed
make[2]: *** [CMakeFiles/vcompile.dir/all] Error 2
CMakeFiles/Makefile2:175: recipe for target 'CMakeFiles/vcompile.dir/rule' failed
make[1]: *** [CMakeFiles/vcompile.dir/rule] Error 2
Makefile:168: recipe for target 'vcompile' failed
make: *** [vcompile] Error 2

-----------END-----

Any idea what can be the solution for this?

gautschimi, 2017-04-15:
since you are using a 32b modelsim version it obviously cannot run in 
64b mode. hence, you have to remove the "-64" option in the file 
sw/apps/CMakeSim.txt and reexecute the configure file.

btw the -m32 is a flag for the riscv compiler and has nothing to do with 
modelsim. are you sure that you are using our compiler? you can find it 
here:

https://github.com/pulp-platform/ri5cy_gnu_toolchain

Aregaard, 2017-04-15:
OK thank you, It took away the -m32 error. I used the RISCV-toolchain newlib before. 
I removed all the -64 in the sw/apps/CMakeSim.txt. and reconfigure the ./cmake_configure.riscv.gcc.sh But it still complains about 

"Failed to open executable /home/aregaard/intelFPGA/16.1/modelsim_ase/linuxaloem/../linux_x86_64pe/vlog in execute mode needed for the option -64."

"CMakeFiles/vcompile.dir/build.make:57: recipe for target 'CMakeFiles/vcompile' failed
make[3]: *** [CMakeFiles/vcompile] Error 1
CMakeFiles/Makefile2:168: recipe for target 'CMakeFiles/vcompile.dir/all' failed
make[2]: *** [CMakeFiles/vcompile.dir/all] Error 2
CMakeFiles/Makefile2:175: recipe for target 'CMakeFiles/vcompile.dir/rule' failed
make[1]: *** [CMakeFiles/vcompile.dir/rule] Error 2
Makefile:168: recipe for target 'vcompile' failed
make: *** [vcompile] Error 2"

Shall I try to install ubuntu 32-bit instead of 64-bit and start over? Or is it the Modelsim that needs to be in 64-bit?

gautschimi, 2017-04-15:
It is modelsim that complains. It has nothing to do with ubuntu.
I would delete the build folder and check where else the -64 is set. Probably also in one of the scripts in pulpino/vsim/tcl_files/*
Then regenerate the build foldet and reconfigure and everything should work.

Best
Michael

Aregaard, 2017-04-15:
Thank you! Almost all of the tcl files had -64 flags and vcompile .sh-files.
Deleted the build folder and created a new one with a new copy of cmake_configure.riscv.gcc.sh

"Build target compile"

Best Regards and thank you for quick response to questions

closed 2017-04-19 by Aregaard