pulp-platform/pulpino #66: Help with helloworld.vsim

javiermacotela, 2017-04-16:
Hello I have been assigned a school project using pulpino, after I install the correct compiler and run make vcompile I try to run make helloworld.vsim, and I get this error: 

[  0%] Building C object libs/bench_lib/CMakeFiles/bench.dir/src/bench.c.o
/home/javier/icproject/pulpino/sw/libs/sys_lib/inc/utils.h: Assembler messages:
/home/javier/icproject/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/home/javier/icproject/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/home/javier/icproject/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/home/javier/icproject/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/home/javier/icproject/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/home/javier/icproject/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/home/javier/icproject/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/home/javier/icproject/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/home/javier/icproject/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/home/javier/icproject/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/home/javier/icproject/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/home/javier/icproject/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
/home/javier/icproject/pulpino/sw/libs/sys_lib/inc/utils.h:81: Error: no such instruction: `l.mfspr %eax,%eax,0'
libs/bench_lib/CMakeFiles/bench.dir/build.make:62: recipe for target 'libs/bench_lib/CMakeFiles/bench.dir/src/bench.c.o' failed
make[3]: *** [libs/bench_lib/CMakeFiles/bench.dir/src/bench.c.o] Error 1
CMakeFiles/Makefile2:528: recipe for target 'libs/bench_lib/CMakeFiles/bench.dir/all' failed
make[2]: *** [libs/bench_lib/CMakeFiles/bench.dir/all] Error 2
CMakeFiles/Makefile2:1750: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule' failed

Could I get some guidance on how to fix this? Thanks

be4web, 2017-04-16:
Hi,

it seems that your are using the wrong compiler e.g.: it doesn't recognize the __riscv__ define. As far as I know this is how the recent gcc behaves. So you basically have 4 options:

1. define __riscv__
2. change it to a define which is set (don't nail me down on this but this might be something like ___riscv32__ or __riscv64__
3. use our compiler https://github.com/pulp-platform/ri5cy_gnu_toolchain/blob/master/README.md
4. use an older official compiler

Personally I would go for 3.

Cheers

Florian 

Mit freundlichen Grüßen 
Florian Zaruba

javiermacotela, 2017-04-16:
Hello, yes it was precisely that, I fixed it and it is working now, thanks a lot!

closed 2017-04-16 by javiermacotela

suhasskrishanmurthy, 2019-07-02:
@javiermacotela did you try building freertos application because i am trying making above mentioned modification but am error like this 
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c: Assembler messages:
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:232: Error: no such instruction: `lw sp,pxCurrentTCB'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:233: Error: no such instruction: `lw sp,(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:234: Error: no such instruction: `lw t0,0x00(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:235: Error: no such instruction: `csrw mepc,t0'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:236: Error: no such instruction: `lw x1,0x04(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:237: Error: no such instruction: `lw x3,0x08(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:238: Error: no such instruction: `lw x4,0x0c(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:239: Error: no such instruction: `lw x5,0x10(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:240: Error: no such instruction: `lw x6,0x14(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:241: Error: no such instruction: `lw x7,0x18(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:242: Error: no such instruction: `lw x8,0x1c(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:243: Error: no such instruction: `lw x9,0x20(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:244: Error: no such instruction: `lw x10,0x24(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:245: Error: no such instruction: `lw x11,0x28(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:246: Error: no such instruction: `lw x12,0x2c(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:247: Error: no such instruction: `lw x13,0x30(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:248: Error: no such instruction: `lw x14,0x34(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:249: Error: no such instruction: `lw x15,0x38(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:250: Error: no such instruction: `lw x16,0x3c(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:251: Error: no such instruction: `lw x17,0x40(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:252: Error: no such instruction: `lw x18,0x44(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:253: Error: no such instruction: `lw x19,0x48(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:254: Error: no such instruction: `lw x20,0x4C(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:255: Error: no such instruction: `lw x21,0x50(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:256: Error: no such instruction: `lw x22,0x54(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:257: Error: no such instruction: `lw x23,0x58(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:258: Error: no such instruction: `lw x24,0x5C(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:259: Error: no such instruction: `lw x25,0x60(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:260: Error: no such instruction: `lw x26,0x64(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:261: Error: no such instruction: `lw x27,0x68(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:262: Error: no such instruction: `lw x28,0x6C(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:263: Error: no such instruction: `lw x29,0x70(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:264: Error: no such instruction: `lw x30,0x74(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:265: Error: no such instruction: `lw x31,0x78(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:266: Error: no such instruction: `addi sp,sp,0x7c'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:255: Error: no such instruction: `addi sp,sp,-0x6c'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:256: Error: no such instruction: `sw x1,0x04(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:257: Error: no such instruction: `sw x3,0x08(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:258: Error: no such instruction: `sw x4,0x0c(sp)'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:259: Error: no such instruction: `sw x5,0x10(sp)'
[... cut ...]
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:286: Error: no such instruction: `csrr %eax,mcause'
/home/riscv/pulpino/pulpino/sw/apps/freertos/Source/portable/GCC/RI5CY/port.c:299: Error: no such instruction: `mret'
make[3]: *** [CMakeFiles/freertos.dir/build.make:115: CMakeFiles/freertos.dir/Source/portable/GCC/RI5CY/port.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:73: CMakeFiles/freertos.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:85: CMakeFiles/freertos.dir/rule] Error 2
make: *** [Makefile:118: freertos] Error 2

any idea why?