pulp-platform/pulpino #68: Issue trying to run helloworld.vsim

jgancedo, 2017-04-25:
Hello,

I am going through the same process as fellow mates in these issues:
https://github.com/pulp-platform/pulpino/issues/66
https://github.com/pulp-platform/pulpino/issues/65

So far I've been able to run the whole method until the **make helloworld.vsim** command. An error I haven't been able to identify prevents the simulation to start. This is the output I receive when running the command:
```
Scanning dependencies of target bench
[  0%] Building C object libs/bench_lib/CMakeFiles/bench.dir/src/bench.c.o
[  0%] Linking C static library libbench.a
[  0%] Built target bench
Scanning dependencies of target crt0
[  0%] Building C object CMakeFiles/crt0.dir/ref/crt0.riscv.S.o
[  0%] Built target crt0
Scanning dependencies of target string
[  0%] Building C object libs/string_lib/CMakeFiles/string.dir/src/qprintf.c.o
[  0%] Linking C static library libstring.a
[  0%] Built target string
Scanning dependencies of target sys
[  0%] Building C object libs/sys_lib/CMakeFiles/sys.dir/src/exceptions.c.o
[  0%] Building C object libs/sys_lib/CMakeFiles/sys.dir/src/gpio.c.o
[  0%] Building C object libs/sys_lib/CMakeFiles/sys.dir/src/int.c.o
[  0%] Building C object libs/sys_lib/CMakeFiles/sys.dir/src/spi.c.o
[  0%] Building C object libs/sys_lib/CMakeFiles/sys.dir/src/timer.c.o
[  0%] Building C object libs/sys_lib/CMakeFiles/sys.dir/src/uart.c.o
[  0%] Building C object libs/sys_lib/CMakeFiles/sys.dir/src/utils.c.o
[  0%] Building C object libs/sys_lib/CMakeFiles/sys.dir/src/i2c.c.o
[  0%] Linking C static library libsys.a
[  0%] Built target sys
Scanning dependencies of target Arduino_core
[  0%] Building C object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/wiring_digital.c.o
[  0%] Building C object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/WInterrupts.c.o
[  0%] Building C object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/hooks.c.o
[  0%] Building C object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/wiring.c.o
[  0%] Building C object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/wiring_analog.c.o
[  0%] Building CXX object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/WString.cpp.o
[  0%] Building CXX object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/Print.cpp.o
[  0%] Building CXX object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/Stream.cpp.o
[  0%] Building CXX object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/HardwareSerial.cpp.o
[  0%] Building CXX object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/HardwareSerial0.cpp.o
[  0%] Building CXX object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/abi.cpp.o
[  0%] Building CXX object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/new.cpp.o
[  0%] Building CXX object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/WMath.cpp.o
[  0%] Building C object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/wiring_shift.c.o
[  0%] Building C object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/wiring_pulse.c.o
[  0%] Building CXX object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/src/IPAddress.cpp.o
[  0%] Building CXX object libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/inc/main.cpp.o
[ 50%] Linking CXX static library libArduino_core.a
[ 50%] Built target Arduino_core
Scanning dependencies of target Arduino_separate
[ 50%] Building C object libs/Arduino_lib/separate_libs/CMakeFiles/Arduino_separate.dir/src/twi.c.o
/home/dogtag/projects/pulpino/sw/libs/Arduino_lib/separate_libs/src/twi.c: In function 'ISR_I2C':
/home/dogtag/projects/pulpino/sw/libs/Arduino_lib/separate_libs/src/twi.c:362:6: warning: suggest explicit braces to avoid ambiguous 'else' [-Wparentheses]
    if(twi_state == TWI_MTX)
      ^
[ 50%] Building CXX object libs/Arduino_lib/separate_libs/CMakeFiles/Arduino_separate.dir/src/Wire.cpp.o
[ 50%] Building CXX object libs/Arduino_lib/separate_libs/CMakeFiles/Arduino_separate.dir/src/SPI.cpp.o
[ 50%] Linking CXX static library libArduino_separate.a
[ 50%] Built target Arduino_separate
Scanning dependencies of target helloworld.elf
[ 50%] Building C object apps/helloworld/CMakeFiles/helloworld.elf.dir/helloworld.c.o
[ 50%] Linking CXX executable helloworld.elf
[ 50%] Built target helloworld.elf
Scanning dependencies of target helloworld.stim.txt
[ 50%] Generating helloworld.s19
/bin/sh: 1: --srec-len: not found
apps/helloworld/CMakeFiles/helloworld.stim.txt.dir/build.make:64: recipe for target 'apps/helloworld/helloworld.s19' failed
make[3]: *** [apps/helloworld/helloworld.s19] Error 127
CMakeFiles/Makefile2:1787: recipe for target 'apps/helloworld/CMakeFiles/helloworld.stim.txt.dir/all' failed
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.stim.txt.dir/all] Error 2
CMakeFiles/Makefile2:1762: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule' failed
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
Makefile:675: recipe for target 'helloworld.vsim' failed
make: *** [helloworld.vsim] Error 2
```

Thank you in advance.

gautschimi, 2017-04-25:
have you used our compiler? it seems like the objcopy is not recognizing the option --srec-len
it could also be that your cmake is too old. what version are you using?

jgancedo, 2017-04-25:
My cmake is version 3.5.2.
The compiler I have assigned to environment variable "CC" is "riscv32-unknown-elf-gcc-5.2.0".

jgancedo, 2017-04-25:
Also yes, I used the compiler from https://github.com/pulp-platform/ri5cy_gnu_toolchain

gautschimi, 2017-06-21:
the option --srec-len is used by objcopy
so please check if objcopy is known. "which riscv32-unknown-elf-objcopy" should point to the bin folder of the installed compiler.

closed 2017-08-12 by be4web

venkatashok1234, 2020-08-03:
hi everyone ,
can anyone go through the below error and please let me know

Scanning dependencies of target bench
[  0%] Building C object libs/bench_lib/CMakeFiles/bench.dir/src/bench.c.o
[  0%] Linking C static library libbench.a
[  0%] Built target bench
Scanning dependencies of target crt0
[  0%] Building C object CMakeFiles/crt0.dir/ref/crt0.riscv.S.o
/root/pulpino/sw/ref/crt0.riscv.S:17:3: error: expected identifier or '(' before '.' token
   .section .text
   ^
/root/pulpino/sw/ref/crt0.riscv.S:84:3: error: unknown type name 'addi'
   addi x10, x0, 0
   ^
/root/pulpino/sw/ref/crt0.riscv.S:84:17: error: expected identifier or '(' before numeric constant
   addi x10, x0, 0
                 ^
/root/pulpino/sw/ref/crt0.riscv.S:89:3: error: unknown type name 'mv'
   mv a0, s0
   ^
/root/pulpino/sw/ref/crt0.riscv.S:91:3: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'jal'
   jal  x1, exit
   ^
/root/pulpino/sw/ref/crt0.riscv.S:254:5: error: invalid preprocessing directive #These
   # These don't have to do anything since we use init_array/fini_array.
     ^
/root/pulpino/sw/ref/crt0.riscv.S:254:14: warning: missing terminating ' character
   # These don't have to do anything since we use init_array/fini_array.
              ^
/root/pulpino/sw/ref/crt0.riscv.S:265:3: error: expected identifier or '(' before '.' token
   .org 0x00
   ^
CMakeFiles/crt0.dir/build.make:81: recipe for target 'CMakeFiles/crt0.dir/ref/crt0.riscv.S.o' failed
make[3]: *** [CMakeFiles/crt0.dir/ref/crt0.riscv.S.o] Error 1
CMakeFiles/Makefile2:3980: recipe for target 'CMakeFiles/crt0.dir/all' failed
make[2]: *** [CMakeFiles/crt0.dir/all] Error 2
CMakeFiles/Makefile2:5030: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule' failed
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
Makefile:694: recipe for target 'helloworld.vsim' failed
make: *** [helloworld.vsim] Error 2