pulp-platform/pulpino #70: Integrate a hardware accelerator in the architecture

jgancedo, 2017-05-09:
Hello,

We are a group of students trying to integrate a custom hardware accelerator into the PULPino. We have read the official documentation, but yet cannot find the exact method or guide on how to:

- Map certain memory addresses to our hardware accelerator in order to be visible from APB and AXI buses.
- Create our own simulation running C code on the PULPino to access the accelerator.

We would be grateful if you could point us to some guide we may have missed or just in the right direction.

gautschimi, 2017-05-09:
Hi there

--------------------------------------
1. how to map a hardware accelerator to apb or axi:
Depends a bit what kind of accelerator you have. if it's a slave I'd 
suggest to look at the existing apb peripherals, and extend it with an 
additional port.
(see pulpino/rtl/peripherals.sv)

so you have to instantiate a new bus s_hwacc_bus() like s_timer_bus() 
and connect it on one side to your accelerator, and on the other side to 
the periph_bus_wrap which has one slave port (s_apb_bus) and several 
master ports. the slave port is converted to an axi bus (in axi2apb) and 
then attached to the axi_interconnect (in pulpino_top.sv)
(no change need on axi bus in this case)

---
if you want to attach it to the axi interconnect, you have to add 
another slave or master port at the axi_interconnect (in pulpino_top.sv) 
by changing the parameters NB_MASTER or NB_SLAVE which are currently 
both at 3.
then you have to increase the dimensionality of the axi bus, (slaves, or 
masters) and connect slaves[4] or masters[4] to your hardware accelerator

I'd first study the two protocols (axi, apb) a bit, then decide what 
suits your accelerator better, and then implement it

--------------------------------------
2. how to access it in C:

it's quite simple in C. just define an address and cast it to a pointer.

// define the address of your accelerator
#define HWACC_BASEADDR 0x10000000

// write to this address
*(volatile int*)HWACC_BASEADDR = some_value;

// read from this register
state = *(volatile int*)HWACC_BASEADDR;
--------------------------------------

hope it helps
best,

Michael

jgancedo, 2017-05-09:
Hello @gautschimi,

Thank you for your answer. We will go through it. If it is ok, we may leave this issue open, as we will probably come about other problems in the process.

Thanks again for your help.

Aregaard, 2017-05-12:
How do we increase the dimensionality of the AXI-bus?
We changed the size of NB_SLAVE and NB_MASTER to 4 and added the memory for it
(according to the memory map from the datasheet). We get an error in Modelsim: "axi_node_intf_wrap.sv(28): The interface array port 'slave' of size 4 must be passed an identical array."

gautschimi, 2017-05-12:
the slave interface is instantiated in pulpino_top.sv and has a 
dimensionality 3 (2:0), so just increase it to slaves[3:0], and use 
slaves[3] for your accelerator

==>

AXI_BUS
   #(
     .AXI_ADDR_WIDTH ( `AXI_ADDR_WIDTH     ),
     .AXI_DATA_WIDTH ( `AXI_DATA_WIDTH     ),
     .AXI_ID_WIDTH   ( `AXI_ID_SLAVE_WIDTH ),
     .AXI_USER_WIDTH ( `AXI_USER_WIDTH     )
   )
   slaves[3:0]();

jgancedo, 2017-06-07:
Hello @gautschimi, 

So far we have been able to integrate our own accelerator with its AXI master in the architecture. Now we have almost integrated a RAM acting as an AXI slave, but we come up with an error we can't figure out:
/rtl/axi_node_intf_wrap.sv(195): Variable '/tb/top_i/slaves[3]/b_ready' written by more than one continuous assignment. See /ips/matrix_mult/axi_ram_wrap.sv(86)

Where axi_ram_wrap.sv is the wrapper of our AXI slave. We have checked that our slave doesn't contain multiple assignments to b_ready signal.    The problematic line 195 of axi_node_intf_wrap.sv is:
assign                        master[i].b_ready = s_master_b_ready[i];

We have increased the number of .NB_MASTER in the instantiation of axi_node_intf_wrap in top.sv and also added the AXI slave interface of our block to peripherals. What could we be missing?

gautschimi, 2017-06-08:
Yes it seems that you are writing b_ready multiple times. this is not 
possible as it would lead to drive conflicts. is this the only error you 
have? does it only complain about the b_ready signal, or do you have the 
same error for the other master signals?

did you change anything in this file? what are the parameters you use for:
NB_MASTER
NB_SLAVE
AXI_ADDR_WIDTH
AXI_DATA_WIDTH
AXI_ID_WIDTH
AXI_USER_WIDTH

also check if the dimensionality of master, and s_master_b_ready match.

I'm sorry, I can't say what the issue is just like this. if give me 
access to your code I can have a closer look

best,
Michael

jgancedo, 2017-06-08:
Hello @gautschimi,

Yes, apparently the problem is only in that signal, according to Questasim output.
I have left unchnaged top.sv configuration values:
define AXI_ADDR_WIDTH         32
`define AXI_DATA_WIDTH         32
`define AXI_ID_MASTER_WIDTH     2
`define AXI_ID_SLAVE_WIDTH      4
`define AXI_USER_WIDTH          1

Also in axi_node_intf_wrap instantiation:
NB_MASTER = 4 and NB_SLAVE = 4.

I will also send you an MP with our source code. Thank you for your help so far.

gautschimi, 2017-06-14:
could you solve the problem?

jgancedo, 2017-06-14:
Yes, we could fix this problem. Now our project is over, so this issue can be closed. Thank you for your following up and help throughout :)

closed 2017-06-14 by jgancedo

f2013619, 2018-01-17:
hi @jgancedo ,
 I am also trying to integrate a hardware accelerator to the platform. Can you help me with the source code for the same. Thanks in advance

ravi02011993, 2018-03-06:
hi, @jgancedo @f2013619  I am also trying to integrate. Can you send me the source code?

Thanks in advance