pulp-platform/pulpino #71: FPGA ZedBoard - SSH Connection: Permission Denied

GioAgu17, 2017-05-11:
Hi everybody,

I prepared my Linux VM with the command 'make all' in the fpga directory, and everything seemed to be ok. The SD card was also partitioned and I put all required contents on it. After this, I tried to boot the zedboard, and what it appears is:

IP-Config: Got DHCP answer from 10.79.3.252, my address is 10.79.3.220          
IP-Config: Complete:                                                            
     device=eth0, hwaddr=00:0a:35:00:00:00, ipaddr=10.79.3.220, mask=255.255.254
     host=10.79.3.220, domain=inside.dei.polimi.it, nis-domain=(none)           
     bootserver=0.0.0.0, rootserver=0.0.0.0, rootpath=                          
     nameserver0=10.248.17.11, nameserver1=10.248.17.12                         
ALSA device list:                                                               
  No soundcards found.                                                          
EXT3-fs (mmcblk0p2): error: couldn't mount because of unsupported optional feat)
EXT2-fs (mmcblk0p2): error: couldn't mount because of unsupported optional feat)
EXT4-fs (mmcblk0p2): recovery complete                                          
EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)    
VFS: Mounted root (ext4 filesystem) on device 179:2.                            
devtmpfs: mounted                                                               
Freeing unused kernel memory: 216K (40634000 - 4066a000)                        
EXT4-fs (mmcblk0p2): re-mounted. Opts: data=ordered                             
Starting logging: OK                                                            
Initializing random number generator... random: dd urandom read with 11 bits ofe
done.                                                                           
Starting network...                                                             
ip: RTNETLINK answers: File exists                                              
Starting sshd: OK                                                               
                                                                                
Welcome to Buildroot                                                            
buildroot login: root                                                           
# 

Why I can't see the normal login as it is descripted in the fpga.README? Also why it starts with BuildRoot?

Thank you in advance for your attention.

Giovanni

gautschimi, 2017-05-11:
hm everything looks quite normal. Have you tried to login? what happens 
or what exactly is the problem?

GioAgu17, 2017-05-11:
I want to establish an SSH connection. So I open a new terminal and try this:

user@vivado-pcie:~$ ssh root@10.79.3.220
root@10.79.3.220's password: 
Permission denied, please try again.
root@10.79.3.220's password: 
Permission denied, please try again.
root@10.79.3.220's password: 
Permission denied (publickey,password,keyboard-interactive).

The password is not pulp, nor pulpino. I also tried to change password from the previous terminal but nothing.

What do you mean with login?

Thank you so much

gautschimi, 2017-05-11:
Can you login to the board over uart? I think by default the password 
might not even be active.

but you can login over uart and then change the password with passwd to 
whatever you like. it is then stored on the sdcard. when you change the 
password you might need to first set a correct date, otherwise the ssh 
service might not accept it because it is too old.

GioAgu17, 2017-05-11:
I think I logged in over uart, I mean using the uart connection in the ZedBoard.

I tried to change the date, but still the root refuses my trial.

# date --set "2017-05-11 12:17:00"
Thu May 11 12:17:00 UTC 2017
# date
Thu May 11 12:17:02 UTC 2017
# passwd
Changing password for root
New password: 
Bad password: too weak
Retype password: 
Password for root changed by root
# passwd
Changing password for root
New password: 
Retype password: 
Password for root changed by root

then again:
 ssh root@10.79.3.220
root@10.79.3.220's password: 
Permission denied, please try again.
root@10.79.3.220's password: 
Permission denied, please try again.
root@10.79.3.220's password: 
Permission denied (publickey,password,keyboard-interactive).

gautschimi, 2017-05-11:
Hm weird, I've never experienced something like this. just some ideas 
what is worth to check:

- check that the mac address of the zedboard is unique. (e.g. when you 
use two zedboards they have the same default mac address which might end 
up that you try to login to a wrong board)

- check if you have already added authorized keys, and if yes, make sure 
that the permissions to this file are correct (.ssh/authorized_keys)

- generate a private&public key and copy the public one to the board 
(with ssh-copy-id) like this you will no longer require a password

hope anything of this helps,

best,
Michael

GioAgu17, 2017-05-11:
I copied my public key into the root and now it works!

Thank you so much for the help!

user@vivado-pcie:~/.ssh$ ssh root@10.79.3.220
# 

Best,
Giovanni

closed 2017-05-11 by GioAgu17