pulp-platform/pulpino #73: Queries regarding synthesis of PULPino

nisha9t, 2017-05-16:
Hello,

We are group of students who are working on integrating our hardware accelerator with PULPIno. So far we have able to integrate our HA through APB (with APB slave Wrapper) with PULPino. We tried to run the synthesis for the same on mentor graphics - Design Vision on 65nm and we have few below mentioned queries for the same -

a. At the start, we got some unsynthesizable code in random_stalls.sv file (e.g. case statement). We looked into it and understood it is kind of test code so might not be mandatory to have it. Hence we commented the random_stalls.sv file from our synthesis script.

Is it okay to comment this file and if yes, then is there any tool that can help us to remove or comment the unsynthesizable code from the PULPino instead of doing it manually ?

b. Our synthesis script ran for 9 hours and we got 0.2 micro meter2 as in area.

Can you please confirm if it is expected to run for that long or is there something fishy with our synthesis ?

c. At the end of synthesis we got a warning saying that design contains some black box components. I haven't added the '.sv' files from 'ips' folder where some files get created after running the cmake script.

Is it due to this we are getting the warning for black box components ?

Should we include these files and run the synthesis again or is it something else that we might be missing that is interpreted by the compiler as black box components ?

gautschimi, 2017-05-16:
> Hello,
>
> We are group of students who are working on integrating our hardware
> accelerator with PULPIno. So far we have able to integrate our HA
> through APB (with APB slave Wrapper) with PULPino. We [...]
yes this is a simulation only file, and not used by default.
>
> b. Our synthesis script ran for 9 hours and we got 0.2 micro meter2 as 
> in area.
>
> Can you please confirm if it is expected to run for that long or is 
> there something fishy with our synthesis ?
well the runtime is a bit high, but it depends on your constraints, and 
computer. what is your timing constraint? and what is the max speed you 
achieve? your area is also a bit small, I think you are missing some files.
in a 65nm tech we need ~0.7mm^2 and achieve a 2.5ns clock period in a 
1.08V WC corner

> c. At the end of synthesis we got a warning saying that design
> contains some black box components. I haven't added the '.sv' files
> from 'ips' folder where some files get created after running the [...]

that is not good. the files in the ips folder have to be included of 
course. the only ones you don't need are the ones in rtl/components/* 
have to be replaced with standard cells/ memories from your technology. 
those are just generic files which can be used for simulations.
>
> <https://github.com/pulp-platform/pulpino/issues/73>, or mute the 
> thread

be4web, 2017-05-16:
Hi,

regarding the unsynthesiseable part: There are `ifdef SYNTHESIS around it. You should probably check if your synthesiser sets those correctly or if you need to specify some other macros. 

regarding runtime: 9 hours seems way to high, even for tight constraints. As Michael pointed out I guessing you are trying to synthesise behavioural SRAMs. Replace them with your RAM macros or leave them as a black box (but don't forget to specify appropriate timings on the interface).

If you need some more information you can contact me via email.

Cheers,

Florian

closed 2017-05-16 by be4web

ravi02011993, 2018-03-07:
Hi @nisha9t,
I am integrating a hardware accelerator to APB bus. Could you please share me the source code after the integration. or at least explain the necessary changes to be done?