pulp-platform/pulpino #75: ModelSim compatible version with Pulpino?

GioAgu17, 2017-05-23:
Hi! 

Sorry for bothering. 

I tried to use a version of model Sim but not worked, and it figured out that the only compatible version is Questa x64bit ModelSim, which is not free. 

Is it true that only payment versions of Model Sim are compatible for using PULPino on ZedBoard or is there any free ModelSim version? 

Thank you so much for your attention, 
Giovanni

gautschimi, 2017-05-24:
Hi Giovanni

Just out of the box it might indeed not work with another version. but 
we cannot provide simulation scripts for each and every version. 
However, the code is pure synthesizable system verilog, which can be 
simulated by any simulator support system verilog.

We mostly use 64b versions, but this is not a requirement. If you parse 
through the scripts you will find a lot of "-64" flags. If you remove 
those, it will already run with 32b versions

GioAgu17, 2017-06-04:
Great thank you so much for your kind answer. Finally we arrived to execute something on a zedboard. 
Bestes,
Giovanni

closed 2017-06-04 by GioAgu17