pulp-platform/pulpino #78: Unrecognized command line option '-m32'

manojcs26, 2017-06-08:
Im using the official toolchain provided by RISCV.org. and I am getting this error when I try to run 
**make helloworld.vsim**
.
.
.
_[  0%] Building C object libs/bench_lib/CMakeFiles/bench.dir/src/bench.c.o
riscv32-unknown-elf-gcc: error: unrecognized command line option '-m32'_

gautschimi, 2017-06-08:
please use our derivation of the official toolchain for pulpino:

https://github.com/pulp-platform/ri5cy_gnu_toolchain

manojcs26, 2017-06-08:
Oooh, Thank you

manojcs26, 2017-06-09:
Can I know what is the difference between the official toolchain and the ri5cy toolchain...?
What should be done to run this with the official toolchain(If it can be run)...?

be4web, 2017-06-09:
Yes, you can (with some work) use the newest official one. The problem you experienced is due to the fact that we forked the toolchain a while ago where the -m32 option was still valid. Since then they removed it from the official toolchain. You probably have to investigate a bit in what flags achieve the same behavior. 

Furthermore, our toolchain support the inference of a couple of pulp related instructions. So in general our tc will give you better performance with our core.

In any case, if you have a look at our tc repo you will see that there are just a couple of patches to a particular version of the official toolchain. If you just take this version and build it you will have a version which is for sure compatible.

Cheers
Florian

closed 2017-06-09 by be4web