pulp-platform/pulpino #79: Error when exe "make helloworld.vsim : env:-64 No such file or dir

icdesigner2004, 2017-06-08:
make helloworld.vsim
[  0%] Built target bench
[  0%] Built target crt0
[  0%] Built target string
[100%] Built target sys
[100%] Built target helloworld.elf
[100%] Built target helloworld.stim.txt
[100%] Built target helloworld.links
[100%] Built target helloworld.slm.cmd
[100%] Running helloworld in ModelSim
env: -64: No such file or directory
make[3]: *** [apps/helloworld/CMakeFiles/helloworld.vsim] Error 127
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/all] Error 2
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
make: *** [helloworld.vsim] Error 2

gautschimi, 2017-06-21:
you have to start modelsim in 64b mode. alternatively you can remove the -64 option in the scripts. you can also browse through previous issues, this problem has been discussed already

closed 2017-06-21 by gautschimi

f2013619, 2017-09-13:
Hi, even I got the same issue but I am already using moodelsim 64 bit setup. Can you help me with this?