pulp-platform/pulpino #80: Assertion Error Modelsim

mserrand, 2017-06-09:
Hey Pulpino team,

I am facing problems when changing the memload parameter from "PRELOAD" to "STANDALONE" in the testbench. I successfully changed this parameter from "PRELOAD" to "SPI" to analyze the simulation on Modelsim (with the Helloworld program) but when I am using "STANDALONE" parameter,  I face the following error in modelsim console : 
```
UsingMEMLOAD method: STANDALONE
# [SPI] Enabling QPI mode
# [TRACER] Output filename is: trace_core_00_0.log
# Branch decision is X
# Branch decision is X
# ** Error: Assertion error.
#    Time: 291700 ns Started: 291700 ns  Scope: tb.top_i.core_region_i.RISCV_CORE.load_store_unit_i File: /home/*******/Documents/******/pulpino/vsim/..//ips/riscv/load_store_unit.sv Line: 486
```
I am using Modelsim version 10.2C.
Any hints on how to resolve this problem ?

Thanks a lot !

Maxime

be4web, 2017-06-09:
Hey,

in PRELOAD mode PULPino will boot from its internal boot rom. The boot rom instructs the core to fetch data from an external flash. Unfortunately we can't provide the flash model with our open source release. For that reason PULPino will not succeed in loading its memories and then finally jump to the start of the instruction RAM which only contains 'x's. If you need to use the STANDALONE mode I would suggest you have a look at the boot rom as well as you organise yourself a flash model and instantiate it in the TB.

Hope that helps.
Cheers Florian

mserrand, 2017-06-09:
Thanks for your fast answer,

I now understand why I can't use the Standalone mode. However, the sentence 

> in PRELOAD mode PULPino will boot from its internal boot rom

surprises me. 
Indeed, I saw in simulation that in the Preload mode, the Instruction RAM and Data RAM were simultaneously written (around 18500ns) with the files l2_stim.slm and tdcm_bank0.slm. I don't see what is the role of the bootrom here.

Thanks,

Maxime

be4web, 2017-06-09:
I am sorry, I didn't mean PRELOAD in the first sentence but STANDALONE. Indeed in PRELOAD mode the RAMs are written by the TB. So the boot rom only has a purpose in STANDALONE mode. This mode is mainly to have a test possibility for the chips we do, because there you obviously can't preload your RAMs :-)

mserrand, 2017-06-09:
Great, that's what I thought.
Thanks a lot

closed 2017-06-11 by be4web