pulp-platform/pulpino #82: Demultiplexer PULPino

mserrand, 2017-06-19:
Hi Pulp team,

I'm studying PULPino structure and its SystemVerilog sources and I'm not finding the source of the demultiplexer (see red selection on the joined image). 
[image]

The role of this demultiplexer is to drive requests signals (coming from core)  to RAM Multiplexer or to the Core2AXI component depending of the address. 

Can someone tell me where can I find this demultiplexer ? (or how signals are driven to the correct component)

Thanks a lot

Maxime

gautschimi, 2017-06-21:
Hi Maxime

You can find this unit here:
https://github.com/pulp-platform/pulpino/blob/master/rtl/core_region.sv#L243-L288

it is just checking the outgoing address, masking requests, and then 
forwarding the correct data from the right source.

best,
Michael

mserrand, 2017-06-21:
Thank for your answer, I have other completely different questions.

I thought that the program stored in boot_code.sv was boot_code.c (Argon/pulpino/sw/apps/boot_code). But when i compile this program and compare the slm files generated with the one in boot_code.sv, it is not similar. 

 I also don't know how data and instructions are organized in the bootrom.
 
Thanks for you precisions

Maxime

gautschimi, 2017-06-21:
it is the bootcode. you have to compile it with "make boot_code.install" 
to generate the sv file.

the boot_code is used when the program is stored on a flash which is 
connected over SPI. so in standalone mode, pulpino is first executing 
the bootcode, which accesses the flash over spi and copies the program 
to the instruction memory. since we cannot release a flash model, this 
mode is not supported on the open source pulpino version.

in the open source pulpino you have two options to load the program:
a) initialized directly in the testbench using $memload commands (not 
synthesizable of course, but faster) (make helloworld.vsimc)
b) written over the SPI interface to the instruction memory (slower, but 
more realistic on a real chip)  (make helloworld.vsim.spi)

best,
Michael

mserrand, 2017-06-21:
In fact I would like to use the standalone mode of the testbench with an EEPROM model . I will preload this EEPROM with the helloworld slm files for example.
I have to modify the boot_code.c. You gave me the response I was looking for i.e use make boot_code.install to generate a new boot_code.sv.

Thanks a lot

closed 2017-06-22 by mserrand