pulp-platform/pulpino #84: Execution time of bench algorithms

GioAgu17, 2017-06-23:
Hi,

I have runned on PULPino several algorithms and wanted to know what is the execution time expressed in units of time.

I runned MatrixMul32 and it printed out execution time: 14454. This is the number of clock cycles, right? So I have to multiply this number to the clk time, which is 1/clk_frequency. Clk_frequency of PULPino is 33MHz, right? 

Sorry for these trivial questions but I have to make sure about that.

Also, the number of clock cycles is always the same varying the pc which compiles the program or not?

Finally, when executing bench algorithms such as FFT or IPM, it prints out the number of cycles ( and also other stuff). That number has always to be multiplied to 1/clk_frequency in order to obtain exec time expressed in units of time, right?

Thak you for the attention.
Bestes,
Giovanni

gautschimi, 2017-06-23:
yes it always reports the number of cycles, thus you have to divide this 
number by the frequency

GioAgu17, 2017-06-23:
> Michael Gautschi phone: +41 44 632 99 58 <+41%2044%20632%2099%2058>

Ah ok thanks and for the other questions?

gautschimi, 2017-06-23:
clock frequency of pulpino is how you set it. I think by default it is 
50MHz.

the number of clock cycles does not depend on your computer, only on the 
source code of the program, and the compiler you use.

GioAgu17, 2017-06-23:
Thanks a lot.

In ultimate instance, how can I verify how much is the clock frequency?

Thanks,
Giovanni

gautschimi, 2017-06-23:
I quickly checked. the output frequency of the PS is 50 MHz, and pulpino 
itself has a clocking block which divides the frequncy to 5 MHz by default.

so if you have not changed anything pulpino runs at 5 MHz. in the 
spiload (fpga/sw/apps/spiload/main.c) you can set the two ccr0/ccr2 
registers which control the clock divider. like this you can vary the 
frequency of pulpino ( have a look at the documentation of the clock 
wizard block for that)

https://www.xilinx.com/support/documentation/ip_documentation/clk_wiz/v5_1/pg065-clk-wiz.pdf

(table 4.2 shows what the values of ccr0/ccr2 mean)

ccr0 = C_BASEADDR + 0x200
ccr2 = C_BASEADDR + 0x208

best,
Michael

GioAgu17, 2017-06-24:
Thank you so much 
Giovanni

closed 2017-06-24 by GioAgu17