pulp-platform/pulpino #87: Missing instruction if continuous jr_stall_o assert

kerrmark, 2017-06-27:
Before Jump, if there are more than 1 instruction pending, jr_stall_o will assert twice or more, but jump_done only assert once. that make the last instruction before jump(jar, jalr...) not execute.
[image]
This issue can be fix by change the default value of jump_done:

//code from controller.sv
    pc_mux_o         = PC_BOOT;
    pc_set_o         = 1'b0;
//    jump_done        = jump_done_q;
    **jump_done        = 1'b0;**

davideschiavone, 2017-06-27:
Hi! Thanks for reporting it!

I am not sure I understood it, can you provide us a small assembler code where this situation happens?

Thx

kerrmark, 2017-06-27:
the code are as below:
1ff017d0:       fb010113                addi    sp,sp,-80
1ff017d4:       03512a23                sw      s5,52(sp)
1ff017d8:       00052a83                lw      s5,0(a0)
1ff017dc:       04912223                sw      s1,68(sp)
1ff017e0:       05212023                sw      s2,64(sp)
1ff017e4:       03412c23                sw      s4,56(sp)
1ff017e8:       03612823                sw      s6,48(sp)
1ff017ec:       03712623                sw      s7,44(sp)
1ff017f0:       03812423                sw      s8,40(sp)
1ff017f4:       04112623                sw      ra,76(sp)
1ff017f8:       04812423                sw      s0,72(sp)
1ff017fc:       03312e23                sw      s3,60(sp)
1ff01800:       00caa783                lw      a5,12(s5)
1ff01804:       00050493                mv      s1,a0
1ff01808:       000780e7                jalr    a5
1ff0180c:       01010913                addi    s2,sp,16
1ff01810:       00050c13                mv      s8,a0
1ff01814:       02900b13                li      s6,41 # 29 <HEAP_SIZE-0x1fd7>
1ff01818:       3e700a13                li      s4,999 # 3e7 <HEAP_SIZE-0x1c19>
1ff0181c:       1f400b93                li      s7,500 # 1f4 <HEAP_SIZE-0x1e0c>
1ff01820:       0004a403                lw      s0,0(s1)
1ff01824:       00092023                sw      zero,0(s2)
1ff01828:       00012a23                sw      zero,20(sp)
1ff0182c:       00012c23                sw      zero,24(sp)
1ff01830:       00012e23                sw      zero,28(sp)
1ff01834:       01610823                sb      s6,16(sp)
1ff01838:       00c42783                lw      a5,12(s0)
1ff0183c:       000780e7                jalr    a5
1ff01840:       00050993                mv      s3,a0

for more clearly, i paste a waveform after RTL change as below:
[image]

davideschiavone, 2017-06-28:
Hi,

sorry, like this is not really easy to understand what happened and where.

Maybe we can help you better if you provide the trace of the execution, the jump instruction you are talking about and what is not executed due to this jump.
The more details you give us, the easier it is :)

That would be easier for us to debug and try to reproduce the error.

Thanks a lot for your support,
Davide

timdudu, 2017-06-30:
Hi Davide,
when the core execute follows three instructions, the purpose is jump to the address 0x80 read from 32(s0).

///-------- instruction line ---------------////
lw	a5,16(s0)  --- instruction 1 (read data 0x40 from 0x2000_0010)
lw	a5,32(s0)  --- instruction 2 (read data 0x80 from 0x2000_0020)
jalr	a5         --- instruction 3 (jump to the 0x80 read from 0x2000_0020)

But in some case, it may failed and jump to the address read from 0x2000_0010. After digging into the design, we found the signal jr_stall_o will set when the decode instruction is BRANCH_JALR meanwihle write to the register.

////---- RTL in controller.sv ------////
  if ((jump_in_dec_i == BRANCH_JALR) &&
      (((regfile_we_wb_i == 1'b1) && (reg_d_wb_is_reg_a_i == 1'b1)) ||
       ((regfile_we_ex_i == 1'b1) && (reg_d_ex_is_reg_a_i == 1'b1)) ||
       ((regfile_alu_we_fw_i == 1'b1) && (reg_d_alu_is_reg_a_i == 1'b1))) )
  begin
    jr_stall_o      = 1'b1;
    deassert_we_o     = 1'b1;
  end
////------------- end -----------////

The behavior can be seen in the waveform. 

[image]

Here jr_stall_o rising twice because the instruction 3 is decoded as a jump while the data bus read data from 0x2000_0010(instruction 1) and 0x2000_0020 (instruction 2) to register a5.

when jr_stall_o is set, the core will wait for it to be gone. the design logic is:

////---- RTL in controller.sv ------////
  // if there is a jr stall, wait for it to be gone
  if ((~jr_stall_o) && (~jump_done_q)) begin
    pc_set_o    = 1'b1;
    jump_done   = 1'b1;
  end
////------------- end -----------////

The pc_set_o should be set after the jr_stall_o is down. But from the simulation that the pc_set_o is not set after the second jr_stall_o down. So the core jump to 0x40 read from 0x2000_0010 (instruction 1).

[image]

We try to change the design logic to:

////---- RTL we changed ------////
 if ((~jr_stall_o) && (~jump_done_q)) begin
    pc_set_o    = 1'b1;
    jump_done   = 1'b1;
  end
  else begin
     pc_set_o    = 1'b0;
     jump_done   = 1'b0;               
  end
////------------- end -----------////

The core can jump to the correct address 0x80.

[image]

davideschiavone, 2017-07-03:
Hi timdudu,

I tried the same code internally and it works fine to me.

The difference I noticed between your and mine waveform are in the regfile_we_wb_i that goes down in your case and stays high in mine.

That signal drives the jr_stall that cause you a bug.

Can you investigate more why that signal goes down for you?

Thanks
Davide

timdudu, 2017-07-04:
Hi Davide,

I think that because there's no wait when reading the data RAM in Pulpino platform. The read data is valid immediately in next cycle of gnt. The rvalid of the back to back read stays high. So the regfile_we_wb_i stays high in your case,  as the waveform shows:

[image]

But data accessing in the platform(not Pulpino, I implement RI5CY in another platform) I used may has wait that means the read data may be valid in several cycles after the gnt as described in the datasheet:

[image]

So the rvalid of the back to back read in this situation will not continuous high. The same to regfile_we_wb_i.

[image]

Best Regards,
Tim

davideschiavone, 2017-07-05:
Dear Tim,

Here there is my simulation:

[image]

In RED you see in sequence the REQUEST, GRANT and VALID signal for the first LOAD.
As you see I receive the grant 5 cycles after the request and the valid 5 cycles after the GRANT.

In BLUE you see in sequence the REQUEST, GRANT and VALID signal for the second LOAD.
As you see I receive the grant 5 cycles after the request and the valid 6 cycles after the GRANT.
You may notice that the second REQUEST happens at the same cycle of the first VALID.
Have a look at the LSU finite state machine to see the pipelined operations.

jr_stall_o is high one cycle after the FIRST load receives the GRANT. Indeed the RED GRANT causes the first load to go from the EX stage to the WB stage. Thus the second load goes to the EX stage but the request is not sent yet and the JUMP goes to the ID stage, where they are solved.
It is possible to notice from my waveform that jt_stall_o is always HIGH, so it is a bit strange that in your design it goes down... 

Once finally the second load receives the valid, the cycle after the jump is solved to the correct value.
Indeed the jump does not employ any forward paths as they may be critical in terms of timing (furthermore in case the fwd is coming from data memory, it would create a path DATA to INSTRUCTION memory).

We are making changes in our core and in a month we will update the repository with the new cores.
Btw It would be nice to understand this problem you met.

Cheers,
Davide

timdudu, 2017-07-06:
Hi Davide,

Thanks for your answer.  Finally we found it's not a issue of RI5CY core.  

I have another question about the FENCE instruction support. Does the core support FENCE and  FENCEI instruction ? I found the OPCODE_FENCE in riscv_defines.sv but when executing the instruction it return an illegal instruction exception. 

Thanks,
Tim

davideschiavone, 2017-07-06:
Hi timdudu,

indeed we do not support it :-)

Best,
Davide

closed 2017-08-11 by FrancescoConti