pulp-platform/pulpino #90: Missing load data followed by an conditional branch

timdudu, 2017-07-06:
Hi,

A data load with wait states immediately followed by an conditional branch, the next instruction after the branch instruction might be executed without waiting for the completion of the data load.

///-------- instruction line ---------------////
0x5d18               lw a4,56(a0)     ---------- instruction 1
0xc581               beqz a1, 8(pc)          ---------- instruction 2
0x00e580ab       p.sb a4, 1(a1!)  ---------- instruction 3

instruction 1: Expect the general register a4 to get the value (for example 0x2e4_2023) from the memory address (a0+56), here the data loading from the memory has wait states.
instruction 2: Jump to (pc+8) if a1 = 0. Here the value in a0 is not equal to 0. So pc goes to the next instruction in sequential order.
instruction 3:  Expect to store the value of a4 (0x2e4_2023) get by instruction 1 to the memory address (a1), but 0x2e4_2023 is still on the way to a4 when this instruction is in-executing. It stores an expired value from a4 to the memory.  

waveform:
[image]

The data 0x2e4_2023 with RED border is read from 56(a0) and should be stored to a4 in instruction 1.
The data 0x780_006f with BLUE border is the wrong write data get from a4.

Thanks,
Tim

davideschiavone, 2017-07-07:
Hi Tim,

I see that you receive the GRANT and VALID signals at the same cycle. Which is not the protocol RISCY implements.

It is described in the RISCY documentation, but basically, the valid must happen at least one cycle after grant.

Best,
Davide

timdudu, 2017-07-10:
Hi Davide,

The GRANT and VALID signals at the same cycle are in different transaction. The VALID signal is for the first transaction while the GRANT signal is for the next transaction.

It can be seen in the back to back transaction in RI5CY datasheet.

[image]

Thanks,
Tim

davideschiavone, 2017-07-11:
Hi Tim,

I tried the same test and indeed it looks like a bug.

We are solving it and update the repository soon.

Best,
Davide

timdudu, 2017-07-13:
Hi Davide,

After you solving it,  could you please inform us. So we can test it in our platform.
For convenience,  it's my email: yu_zheng_qiang@126.com .

Thanks,
Tim

closed 2017-10-17 by FrancescoConti